compas-dev/compas_fab #29: Objects discrepancy: compas vs. compas_fab

arpastrana, 2018-11-02:
After installing compas and compas_fab in a conda environment using the instructions provided in the readme, I tried to go on and install compas_fab for Rhino 6.0. However, running `python -m compas_fab.rhino.install 6.0` raises the following error: 

`    from compas.geometry import Frame
ImportError: cannot import name 'Frame'`

I verified the folder structure of the compas library and Frame seems to now be located under `compas.geometry.objects.frame` instead of merely `compas.geometry`. The issue is repeated whenever trying to run any other of the posted examples (i.e. `from compas_fab.backends import RosClient`)

## Expected Result
Successful installation of compas_fab.

## Actual Result
ImportError Exception when calling the core compas library.

## Reproduction Steps
```python
python -m compas_fab.rhino.install 6.0
```

## System Information
OS X 10.13, conda 4.5.11, Python 3.6.6.

gonzalocasas, 2018-11-02:
This issue comes up when you have a pre-release version of either compas or compas_fab (or both) installed or available in the `PYTHONPATH` and you both are very likely to have that situation. Delete the local (old) versions and it will most likely start working. If not, please let me know

arpastrana, 2018-11-05:
I removed manually the `PYTHONPATH` from my `bash_profile` and now I could run 2 or 3 of the compas_fab examples. That makes me think that now is working. However, when trying to install compas_fab for Rhino 6.0, a new error is raised and to my understanding, it has to do with a problem when trying to find the location of the `IronPython` installation folder in Mac. Should I try to manually specify the directory?

I run:
`$ python -m compas_fab.rhino.install 6.0`

And I get:
```
usage: python -m compas_fab.rhino.install [version]

version       Rhino version (5.0 or 6.0)`

Installing COMPAS packages to Rhino IronPython lib:
appdata is None
Traceback (most recent call last):
  File "/Users/arpj/anaconda/envs/deep_timber/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/Users/arpj/anaconda/envs/deep_timber/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/Users/arpj/anaconda/envs/deep_timber/lib/python3.6/site-packages/compas_fab/rhino/install.py", line 62, in <module>
    install(version=version, packages=packages)
  File "/Users/arpj/anaconda/envs/deep_timber/lib/python3.6/site-packages/compas_fab/rhino/install.py", line 37, in install
    compas_rhino.install.install(version, packages)
  File "/Users/arpj/anaconda/envs/deep_timber/lib/python3.6/site-packages/compas_rhino/install.py", line 47, in install
    ipylib_path = compas_rhino._get_ironpython_lib_path(version)
  File "/Users/arpj/anaconda/envs/deep_timber/lib/python3.6/site-packages/compas_rhino/__init__.py", line 48, in _get_ironpython_lib_path
    'settings')
  File "/Users/arpj/anaconda/envs/deep_timber/lib/python3.6/posixpath.py", line 80, in join
    a = os.fspath(a)
TypeError: expected str, bytes or os.PathLike object, not NoneType```

gonzalocasas, 2018-11-06:
For future reference, the PIL error is fixed by downgrading to Pillow 4.0.0:
```
conda install pillow=4.0.0
```

closed 2018-11-06 by gonzalocasas

arpastrana, 2018-11-06:
The issue was solved for me by making symlinks between the compas library in OS X and the IronPython folder in Windows. Thanks, @gonzalocasas.

gonzalocasas, 2018-11-06:
I'll create a follow up on compas' repo to specifically address the scenario of installing compas packages from Mac onto a Parallels Windows setup.

gonzalocasas, 2018-11-06:
https://github.com/compas-dev/compas/issues/173