compas-dev/compas_fab #36: cannot init Robot (compas_fab) with RobotModel from compas

xarthurx, 2019-03-20:
## Expected Result

Init a robot model successfully.

## Actual Result

error report:

``` python
robot model:  Robot name=ur5, Links=7, Joints=6 (6 configurable)
Traceback (most recent call last):
  File "c:/Users/xliot/Documents/_PhD.Research/03_DigitalSculpting/_code/fabScripts/fab.py", line 31, in <module>
    robot = Robot(robot_model)
TypeError: __init__() missing 1 required positional argument: 'robot_artist'

```

## Reproduction Steps

Execute script on the website:
https://gramaziokohler.github.io/compas_fab/latest/examples/02_coordinate_frames.html

## System Information

Operating System name and version, Python version, etc.

xarthurx, 2019-03-20:
I guess this is because the `RobotModel` in `compas` doesn't have the "artist" required for the 'RobotModel' in `compas_fab`.

romanarust, 2019-03-21:
Thanks for pointing out, will soon be fixed with merging from [`robot_refactor` branch](https://github.com/gramaziokohler/compas_fab/blob/robot_refactor/src/compas_fab/robots/robot.py), having `artist=None` as default.

romanarust, 2019-04-23:
Fixed with f850ffc

closed 2019-04-23 by romanarust