compas-dev/compas_fab #39: [suggestion] examples on control a robot via MoveIt

xarthurx, 2019-04-14:
Summary.

MoveIt has a tutorial example on how to operate a simulated robot:
http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/move_group_python_interface/move_group_python_interface_tutorial.html

For now, all the examples and tutorials on the main website of `compas_fab` are about how to setup the library, URDF. 

And functions like `find_path_plan` provided with the VREP example are not available in the `RosClient` class.

Users like me are quite confused about what to do after setup ROS.
IMHO, examples similar to the *MoveIt* ones are super helpful for new users to get started with the lib.

Of if I may ask, are these things implemented yet or still under development? And how mature/stable are the state of `RosClient()` for production use?

Thanks.

gonzalocasas, 2019-04-14:
More examples will come soon-ish. `RosClient` is pretty mature in itself, and it basically allows you to send and receive anything, i.e. it's basic API takes a dictionary (or dictionary-like object) and passes it down to ROS, as long as the fields match, the message will get through. Same thing with the response. So, on that lower-level, anything is callable and usable as is. For MoveIt, there are lots of methods that ease its usage, creating higher level abstractions, classes representing its more common message types and so on; but any topic, service or action is usable without them as well.

gonzalocasas, 2019-04-23:
Added documentation and examples on how to [configure ROS](https://gramaziokohler.github.io/compas_fab/latest/backends/ros.html) and [how to use it for planning via moveit](https://gramaziokohler.github.io/compas_fab/latest/examples/03_backends_ros.html).

closed 2019-04-23 by gonzalocasas