compas-dev/compas_fab #59: [feature/choreo] converting compas.robot into pybullet robot

yijiangh, 2019-08-07:
The proposed function will convert the `compas.robots.RobotModel` [1] to a pybullet body produced by `pybullet.loadURDF`, see for example `load_pybullet` [2].

This is required by pychoreo since it is using pybullet as a collision checker backend.

This is a part of the larger efforts compas-choreo integration #55 .

[1] https://compas-dev.github.io/main/api/generated/compas.robots.RobotModel.html?highlight=compas%20robots%20robotmodel#compas.robots.RobotModel
[2] https://github.com/caelan/ss-pybullet/blob/b89756d0aaa5f8015cd3a59f85efab30a0ab42a1/pybullet_tools/utils.py#L254

yijiangh, 2019-08-07:
This will help us to have a direct way to generate pybullet robot object (instead of tweaking the mesh file paths to relative paths in the URDF). 

It will also make our life much easier for testing the ikfast_pybind [1] modules in a pybullet environment, for example, the test modules here [2].

[1] https://github.com/yijiangh/ikfast_pybind
[2] https://github.com/yijiangh/conrob_pybullet/tree/655a32fa2d768ce810b3fb700abc724f034308ab/debug_examples

yijiangh, 2019-08-07:
a6145d5 "fixes" this issue by having a temporal workaround: saving a temporal URDF file with relative linking paths, and send this temporal URDF to pybullet.

An ideal solution would be `compas_fab.robot.Robot` can generate an urdf package (see the placeholder [`generate_robot_package`](https://github.com/yijiangh/compas_fab/blob/a6145d511b6ce699c8047560ac18550413306dfc/src/compas_fab/robots/robot.py#L1152) and its docstring), but this might involve some changes on the `compas.files.xml` functionalities (for IronPython compatibility).

I'll keep this issue open until the `generate_robot_package` function is implemented.

gonzalocasas, 2020-08-24:
@yijiangh this is now covered with @beverlylytle 's pybullet integration, right? Or is there anything missing?

beverlylytle, 2020-08-24:
As it is now, a pybullet robot can be created from a robot only if there is a urdf attached to it, but robots not loaded from a urdf are not easily imported to pybullet.  It should be possible to make this easy for the user by writing a method to step through a robot's model to create a pybullet robot without going through a urdf, but it might be the case that such an object would have its concave bits convexified in the pybullet simulation.

yijiangh, 2021-05-19:
With compas_fab `0.18.0`, I think this issue can be closed.

closed 2021-05-19 by yijiangh