compas-dev/compas_fab #60: [feature] get_collision_objects method in ros.client class

yijiangh, 2019-08-07:
It would be great if we can have a "fetch back" all the collision objects' meshes from the current planning scene. This kind of functionality is needed when:
- we want to visualize the current status of the planning scenes on their side (see discussions in #32)
- we want to use a planner that does not directly communicate with the moveit::planning_scene. For example, in the case of `pychoreo` (#55), the planner relies on pybullet as a collision checking engine. When we call pychoreo from compas_fab, we will need to first fetch back the collision objects' transformation and meshes, and send them to pychoreo.

gonzalocasas, 2021-02-23:
@yijiangh I think this is already possible via `robot.client.get_planning_scene()`. Should we close?

yijiangh, 2021-02-23:
Yes! It's been a while since I created this issue :)

closed 2021-02-23 by yijiangh