compas-dev/compas_fab #63: [docs] more info on ServiceDescription and how to make ROS service calls

yijiangh, 2019-08-08:
## Code commit
This issue is related to the latest change (https://github.com/compas-dev/compas_fab/commit/bdc23a1178d3772f8672eb378df3afd4046fc206) by @gonzalocasas.

## what do I want

I'm wondering how do we replicate the following simple service call in ROS C++ using the `ServiceDescription` class in `compas_fab.backends.ros`:

```C++
  // fetch scene
  ros::NodeHandle nh;
  auto planning_scene_fetch_client = nh.serviceClient<moveit_msgs::GetPlanningScene>("/get_planning_scene");

  if(!planning_scene_fetch_client.waitForExistence())
  {
    ROS_ERROR_STREAM("Cannot connect with get_planning_scene server...");
  }

  moveit_msgs::GetPlanningScene fetch_srv;
  fetch_srv.request.components.components = moveit_msgs::PlanningSceneComponents::WORLD_OBJECT_GEOMETRY;

  if(!planning_scene_fetch_client.call(fetch_srv))
  {
    ROS_ERROR_STREAM("[Process Planning] Failed to fetch planning scene srv!");
  }

  for(auto& existing_obj : fetch_srv.response.scene.world.collision_objects)
  {
     // Do something
  }

```

Currently, the documentation on the `ServiceDescription` is very sparse. And based on all the service calls that exist in the current codebase, I feel like this class is intended to be the main port from which the users can make service calls to the ROS backend.

By raising this issue, I want to clarify some of my questions in regard to the `ServiceDescription` and how in general a user would make a ROS service call through `compas_fab`.

## My attempts

Mimicking the existing service calls (`GetPositionIK`, `GetMotionPlan` etc.), I try to make a `get_planning_scene` service call in the following way:

inside the `MoveItPlanner(PlannerPlugin)`, we define the service description
```python
class MoveItPlanner(PlannerPlugin):
    ...
    GET_PLANNING_SCENE = ServiceDescription('/get_planning_scene',
                                            'GetPlanningScene',
                                            GetPlanningSceneRequest,
                                            GetPlanningSceneResponse,
                                            validate_response)
    
    def xxx(self):
    ...
```

and then we can define the function that can make the service call:

```python
# MoveItPlanner class function
def _get_collision_objects(self)
        request = dict(components=PlanningSceneComponents.WORLD_OBJECT_GEOMETRY)
        planning_scene_reponse = GetPlanningSceneResponse()

        def err_fn(err_msg):
            print(err_msg)
            assert False

        self.GET_PLANNING_SCENE(self, request, planning_scene_reponse, err_fn)
        print(planning_scene_reponse.scene)
```

Finally, the user makes the service call:

```python
from compas.datastructures import Mesh

import compas_fab
from compas_fab.robots import Robot as RobotClass
from compas_fab.robots.ur5 import Robot
from compas_fab.robots import CollisionMesh
from compas_fab.backends import RosClient
from compas_fab.robots import PlanningScene

def test_get_collision_objects_from_planning_scene():
    with RosClient() as client:
        assert client.is_connected
        robot = Robot(client)

        scene = PlanningScene(robot)
        mesh = Mesh.from_stl(compas_fab.get('planning_scene/floor.stl'))
        cm = CollisionMesh(mesh, 'floor')
        scene.add_collision_mesh(cm)

        collision_objs = client._get_collision_objects()
        print(collision_objs)
```

And the terminal gives me `[]`, but apparently the object `floor` is in the scene.

## Questions

Specifically, I have the following questions:
1. When initializing the `ServiceDescription` class, we have to provide:
``` python
class ServiceDescription(object):
    def __init__(self, name, service_type, request_class=None, response_class=None, validator=None):
     ....
```
We call it by
```python
MoveitPlanner.GET_PLANNING_SCENE(self, request, callback, errback)
```
What is the difference between `request_class`, `reponse_class`, 'validator' and `request`, `callback`, 'errback'?

2. How exactly does `client._get_collision_objects()` activate the `ServiceDescription.call`? Notice that `RosClient` does not even have a function called `_get_collision_objects` and I'm not using `await_callback` to call `_get_collision_objects`.

3. In `MoveitPlanner.GET_PLANNING_SCENE(self, request, callback, errback)`, how does this activate the `ServiceDescription.call`? What does this `self` parameter mean?  

Is it the client in the function argument?
```
# ServiceDescription class function
def call(self, client, request, callback, errback)
```

3.  How do we achieve the check call success as in C++?
```C++
if(!planning_scene_fetch_client.waitForExistence())
// OR
if(!planning_scene_fetch_client.call(fetch_srv))
```

gonzalocasas, 2019-08-08:
The planning scene retrieval was not fully implemented, but I just added the missing bits on the branch so that you can give it a go.

For completeness/reference purposes, `ServiceDescription` is an implementation detail, it's not intended to be used from outside the library, and its only purpose it so minimize code duplication for the very repetitive handling of service requests.

Your code snippet should now work, by changing the last two lines to:
```
        planning_scene = client.get_planning_scene()
        print(planning_scene)
```

but keep in mind the following:

* `add_collision_mesh` is a topic, so the call returns before the add operation is completed, handling in synchronously would involve monitoring that the object has appeared on the scene and only then unblocking, but that's not implemented at the moment, so, you will probably need to add an arbitrary wait in between.
* `waitForExistence` (waits for the service itself, not for the result of a service, does not currently exist in roslibpy)

> This issue will auto-close once the branch above is merged into master.

yijiangh, 2019-08-08:
@gonzalocasas Thank you very much!

Just to make sure that I understand it correctly, there are two reasons that the following code snippet ([link](https://github.com/compas-dev/compas_fab/commit/ef8efc6db2f415e8495e2801e430a3e1c7ec72dd#diff-ee4d93f8d2c00f0e38e1c1d4c3e1d7cdL304)) before your latest commit https://github.com/compas-dev/compas_fab/commit/ef8efc6db2f415e8495e2801e430a3e1c7ec72dd is not working properly:

```python
    def get_planning_scene(self, callback, components):
        """
        """
        reqmsg = GetPlanningSceneRequest(PlanningSceneComponents(components))

        def receive_message(msg):
            response = GetPlanningSceneResponse.from_msg(msg)
            callback(response)

        srv = Service(self, '/get_planning_scene',
                      'moveit_msgs/GetPlanningScene')
        request = ServiceRequest(reqmsg.msg)
        srv.call(request, receive_message, receive_message)
```

1. The code above is not waiting for the asynchronous service call to complete. The correct way to do any service call that involves fetching parameters/objects back should do it with `await_callback`, for example, https://github.com/compas-dev/compas_fab/commit/ef8efc6db2f415e8495e2801e430a3e1c7ec72dd#diff-ee4d93f8d2c00f0e38e1c1d4c3e1d7cdR204:

```python
    def get_planning_scene(self):
        kwargs = {}
        kwargs['errback_name'] = 'errback'

        return await_callback(self.get_planning_scene_async, **kwargs)

    def get_planning_scene_async(self, *args, **kwargs):
        raise NotImplementedError('No planner plugin assigned')

```

2. Before commit https://github.com/compas-dev/compas_fab/commit/ef8efc6db2f415e8495e2801e430a3e1c7ec72dd, many messages inside the planning_scene is not fully implemented, which will simply give me empty/default values.

-----

So to conclude on how in general a user will make a ROS service call in compas_fab, he/she should use the `await_callback` function to make the code wait for the completion of the asynchronous service code. 

But **inside the async_function**, it does not matter whether he/she wants to use `ServiceDescription` (which is designed just for the sake of internal code reuse) or simply just do:

```python
srv = Service(client, self.name, self.type)
response_msg = ResponseMessageClass()
srv.call(ServiceRequest(request_msg.msg),
            callback=response_msg, errback=errback)

#################
# a concrete example 
#################
"""
# definition of GetPlanningScene.srv
# from: http://docs.ros.org/api/moveit_msgs/html/srv/GetPlanningScene.html

PlanningSceneComponents components
---
PlanningScene scene
"""
srv = Service(self, '/get_planning_scene', 'moveit_msgs/GetPlanningScene')
request_msg = {'components' : PlanningSceneComponents(PlanningSceneComponents.SCENE_SETTINGS)}
response_msg = PlanningScene()
srv.call(request, ServiceRequest(request_msg), receive_msg)
```

Notice that if the `receive_msg` is occupying the `call_back` parameter of the `service.call` function, which makes it non-blocking and the service response will be returned in the callback [1] (in the `receive_msg`).

Please correct me if I'm wrong @gonzalocasas Thank you again!

[1] https://roslibpy.readthedocs.io/en/latest/_modules/roslibpy/core.html#Service.call

yijiangh, 2019-08-08:
Well, I have to admit that most part of my confusion originated from the fact that I didn't read `roslibpy`'s doc thoroughly...

So anyone else who encounters msg / service / action questions in the `compas_fab` context, I'd recommend you to read the doc before posting questions here:
https://roslibpy.readthedocs.io/en/latest/examples.html

gonzalocasas, 2019-08-08:
;) 
Yep, just to confirm, the code didn't work because the parsing of the response was just not fully implemented, so, defaulted in a lot of empty stuff. There was the other smaller issue that adding an object to the scene happens asynchronously but that's a smaller detail.

Regarding the `await_callback`, it's not really needed anymore since `roslibpy 0.5.0`, because it includes that behavior "natively". If you don't pass a callback, the service call behaves synchronously. I could have changed our code in the plugin stuff, but for the sake of consistency with existing code, I chose not to right now. Eventually, I might, but for now, it is not needed.

I hope it's a clearer picture now :)

gonzalocasas, 2019-10-02:
Commit e8fb4d2 addressing this issue is now on master

closed 2019-10-02 by gonzalocasas