compas-dev/compas_fab #85: [suggestion] ROS on conda-forge

jf---, 2019-10-31:
not every day you hear of advances that get you in full tribal dance mode, but I'd say this is a candidate

[ROS on conda-forge](https://medium.com/@wolfv/ros-on-conda-forge-dca6827ac4b6)

romanarust, 2019-10-31:
OMG, awesome!

xarthurx, 2019-10-31:
No windows yet~

jf---, 2019-11-11:
> No windows yet~

Interestingly enough the author mentions working together with [Sean Yen](https://twitter.com/sean_yen), of MS, working on ROS

> "This makes me confident that with his deep knowledge of Windows & ROS, we’ll be able to support Windows for the ROS melodic core packages on conda soon"

Not bad eh ;)

gonzalocasas, 2021-03-29:
I can confirm this actually works, at least with ROS Noetic installing via conda on Windows 10! 🎉

xarthurx, 2021-03-29:
@gonzalocasas  You mean using ROS without WSL?

gonzalocasas, 2021-03-29:
@xarthurx yes! the whole pipeline works now, ros/moveit/rviz, everything running natively on Windows:
[image]

jf---, 2021-06-14:
@gonzalocasas any thoughts on what bar the conda ROS install would need to meet, to be considered a viable alternative to docker?

gonzalocasas, 2021-06-14:
@jf--- actually, it is already very viable. Internally, some projects have been running with that for a couple of months now, we will add docs for it to the main documentation in the short term! ;)

gonzalocasas, 2021-06-14:
I realize that I already said that before, but now we've tested it and are sure it works, I was aiming at adding it to the docs these sprint, but perhaps it slips to the next one in a week or two.

Tobias-Fischer, 2021-06-24:
Hiya - just came across this issue by chance. @wolfv, @traversaro and I would be more than happy to help out if you encounter any issues in running RoboStack, but it seems like things are running smoothly already which is great :)