compas-dev/compas_fab #92: compas_fab.rhino.RobotArtist should support some sort of redraw method.

yck011522, 2019-11-19:
# Feature Request

I want to use `RobotArtist` from `compas_fab.rhino` to visualize my robots.
I want to be able to update the configuration of the robots and redraw the robot.

Current `RobotArtist` do not support drawing to a specific layer, nor support erasing the previous graphics. 

Current `artist.redraw()` does not really redraw things, it is only used to trigger Rhino viewport to redraw. This naming is confusing and inconsistent with many other libries that has a redraw function (Processing, ThreeJS etc). User would expect `redraw()` to be able to either transform the robot to new position or delete old geometry and draw new ones.

## Details

**Describe the solution you'd like**
I want to have a functions such as `artist.clear_layer()`, `artist.redraw()`.
In the case where the robot links or joints have changed, the artist should delete the old robot preview mesh and redraw it entirely.
In the case where the robot configuration changes, the robot preview meshes should be transformed to newly defined config.

gonzalocasas, 2019-11-20:
Fixed/added support for layers and redrawing at part of today's [release v0.10.0](https://gramaziokohler.github.io/compas_fab/latest/changelog.html#0-10-0).

closed 2019-11-20 by gonzalocasas

yck011522, 2019-11-20:
Amazing