compas-dev/compas_fab #96: Robot work envelope visualization

gonzalocasas, 2019-12-06:
# Feature Request

To ease planning tasks and improve understanding of the limits of each robot, it would be desirable to provide access to the 3D work envelope (work space) of a robot model, and allow its visualization via robot artists.

gonzalocasas, 2019-12-06:
http://wiki.ros.org/reuleaux

jf---, 2019-12-06:
The kind of contours you see in `robotstudio` are pretty easy to integrate, can share some code if your interested

gonzalocasas, 2019-12-06:
@jf--- sure! it would be great!

lvasey, 2020-03-06:
Hi all, I am glad this is already a feature request. I intend to  start working on this, implementing what I devleoped in my thesis. I did not know about the ROS reuleaux package, but there are several publications from robotics contexts about different concepts for dexterous workspace defitions :

G. Castelli, E. Ottaviano, and M. Ceccarelli. "A fairly general algorithm
to evaluate workspace characteristics of serial and parallel manipulators."
In: Mechanics based design of structures and machines 36.1
(2008), pp. 14]

J.-P. Merlet. "Geometrical determination of the workspace of a constrained
parallel manipulator". In: Advances in Robot Kinematics, Fer-
rare, Italy, Sept (1992), pp. 7.

T. Lee and D. Yang. "On the evaluation of manipulator workspace".
In: Journal of Mechanisms, Transmissions, and Automation in Design
105.1 (1983), pp. 70. 

 The high level pseudo code I used for a generating a task specific reach envelope specific to a robotic configuration utilizing an IK solver is: 

1. Based on task constraints, determine how the frames should be oriented relative to the robot base and or to different reference geometry (meshes, ect).

2. Strategically reduce the DoF in the system. For example, the first and
last degree axis of a spherical wrist robot in the system can usually be
ignored in the case where the Tool origin is close to the Flange Z-Axis.

3. Generate a system of frames with uniform spacing based on these two
criteria.
4.  Test each frame in each of the unique congurations for reachabilty with
an IK solver. Add the plane it to the solution set if it is reachable and
does not cause a collision.

I would implement this for now and then also think to expand to kinematically redundant robots with a method with FK. Its also important in the case of 6 axis robots that the reach envelope is specific to a IK solution (not all 8 solutions) because in general you can not switch configurations in the middle of a task.