compas-dev/compas_fab #98: Improve cartesian planning options

gonzalocasas, 2020-01-12:


jf---, 2020-01-13:
> Tesseract

Descartes was surpassed by tesseract, which recently got [python bindings](https://github.com/johnwason/tesseract_python). Pilz project is also looking really promising. 
Are you thinking of a generic planner interface, sort of how `moveit` integrates these?

gonzalocasas, 2020-01-13:
yep, we're splitting up all these features (FK, IK, planning, etc) into pluggable `BackendFeature` implementations (see https://github.com/compas-dev/compas_fab/issues/99 for a placeholder issue of that work) with the idea of having generic interfaces for these things (also closely related to your comment of the [kdl solver](https://github.com/compas-dev/compas_fab/issues/13#issuecomment-517757228) ;) )

garyedwards, 2020-08-27:
Hi @gonzalocasas great project, I have just started looking around but have experience with ROS.  What are your current strategies for cartesian velocity objectives (cutting, extruding etc.)?  Have you found workarounds for MoveIt's limitations?

Both [Descartes](https://github.com/PickNikRobotics/descartes_capability) and Pilz have MoveIt plug-ins so can work with compas fab as is.  TrajOpt / Tesseract is defiantly the new and exciting kid on the block also with a [WIP MoveIt plugin](https://github.com/ros-planning/moveit/pull/1626).

I have found that Pilz is all I need most of the time for cartesian planning, unless I need to free up an axis for tight environments.  It has its own action server and python helper library beyond that of MoveIt which adds features like `blend_radius`.  It would be great to expose this in compas fab.  Implementing MoveIt's `set_pose_targets` function could be a quick win as is can be used with the `LIN` planner.

gonzalocasas, 2020-09-05:
@garyedwards thanks! Cartesian planning is still an open point in our roadmap before `1.0` release; for now, we're mostly ignoring the timing information of MoveIt planner and use a different driver for our ABB robots that gives some more control about cartesian velocity, but the goal is to integrate more cartesian planners soon: Descartes, Tesseract, PyChoreo, and Pilz are all in the wish list for integration! :) 

Regarding Pilz in particular, I'm also very keen on integrating it deeper, as it seems to be a very mature project that is tackling the same problems we're facing.

garyedwards, 2020-12-04:
Hi @gonzalocasas now that Pilz is in master extending the compas ROS backed to support the `MotionSequenceRequest` action server and `MotionSequenceRequest` message should give us access to PTP LIN and CIRC sequences.  I need to find some time to look into the compas code further in the new year.  Do you have any pointer or words of warning to avoid any abortive work?  The project seems to be moving fast at the moment.

https://ros-planning.github.io/moveit_tutorials/doc/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html#user-interface-sequence-capability

gonzalocasas, 2020-12-04:
Awesome! The first step towards this would be to prepare a Docker image building master with Pilz (since I believe it's not yet released). Once that is done, probably the best is to implement a new "Backend feature" (Backend features are little classes that add functionality that depends on a Backend, we have a bunch of them in compas_fab for MoveIt, pybullet and creo already as reference)

garyedwards, 2020-12-04:
Perfect I have been playing with my robot in a master branch container for a while with the native ROS tools so that ready for action.  Do you think a new feature or extend the `move_it_plan_motion.py` existing feature?  I guess a `move_it_plan_sequence_motion.py` would be appropriate if new.  It looks like a few new ROS message types also need adding.  I will have a play and share any progress.

gonzalocasas, 2020-12-04:
If it's exposed over a different ROS service, I would probably lean towards a new `move_it_plan_sequence_motion.py` instead of extending the current plan motion one. @beverlylytle what do you think?

> It looks like a few new ROS message types also need adding. I will have a play and share any progress.

Cool 👍

beverlylytle, 2020-12-04:
I agree. ` move_it_plan_sequence_motion.py` ftw!