compas-dev/compas_fab #121: cannot import operations from compas.datastructures.mesh

philianeles, 2020-02-20:
cannot import operations from compas.datastructures.mesh
probably compas_fab version compatibility issue

Steps to reproduce the behavior:
1. Rhino 6, Grasshopper
2. See the image
3. See the image
4. cannot import operations from compas.datastructures.mesh

[image]

 - Windows 10
 - Python 3.7
 - conda

philianeles, 2020-02-20:
It looks like the problem is not there when the compas_fab=0.10.2 instead of compas_fab=0.10.2-c9793b32 (see the image)
[image]

brgcode, 2020-02-20:
it would make sense this is somehow related to `compas_fab` because i have checked release 0.13.3 and everything should be ok with the imports there...

gonzalocasas, 2020-02-21:
I cannot reproduce the problem. I tested the exact same revisions mentioned in the issue, but it worked out fine. The only explanation I can think of is that maybe you had Rhino open and another version of COMPAS was already loaded in memory, when you updated to a different checkout the old version was still in memory and then the imports didn't match.

Since it seems to work for you, and I cannot reproduce, I will go ahead and close, but if you see this again, please re-open or open a new one.

closed 2020-02-21 by gonzalocasas