compas-dev/compas_fab #122: Attach tools for individual planning groups

romanarust, 2020-02-23:
Currently, it is possible to attach a tool to the robot via `robot.attach_tool(tool, group=groupname)` to the end-link of the planning group. However, it only works with one planning group: If I want to attach another tool to another planning group, it get's `detached` from the first planning group. (`robot.attached_tool` gets overridden). Also, `robot.detach_tool()` does not allow to pass a group name.

chen902, 2022-05-06:
Opened https://github.com/compas-dev/compas_fab/pull/356 in an attempt to address this issue by making `Robot.attached_tools` a dictionary which maps a planning group's name to it's attached tool. 

However, the `robot.attach_tool()` also sets the `Tool`s `ToolModel` in the `RobotArtist` associated with this instance. There, multiple tools don't seem to currently be supported. 
In addition, I am not 100% sure if collision detection even supports adding the meshes of multiple tools (or if it even makes sense) and not sure how to test it.

@gonzalocasas @romanarust  maybe you have some input on the above.

gonzalocasas, 2022-09-06:
Done in #356

closed 2022-09-06 by gonzalocasas