compas-dev/compas_fab #124: TypeError: Object of type 'JointConstraint' is not JSON serializable

Qipccc, 2020-03-03:
**Describe the bug**
> when I use parameter "constraints"  in robot.plan_cartesian_motion or robot.inverse_kinematics ,it occured error: "TypeError: Object of type 'JointConstraint' is not JSON serializable"

**To Reproduce**
Steps to reproduce the behavior:
1. Rhino 6,Grasshopper,python3.6, ubuntu1604, ros kinetic, compas 0.11, compas_fab 0.10
2. Sample scripts:
```python
def Robot(client=None,robot_name="ur5", load_geometry=False):
    local_directory = os.path.join(os.path.expanduser('~'), 'robot_description', robot_name) 
    print('local directory: ',local_directory)
    urdf_filename = compas_fab.get(local_directory+'/urdf/robot_description.urdf')
    srdf_filename = compas_fab.get(local_directory+'/robot_description_semantic.srdf')
    model = RobotModel.from_urdf_file(urdf_filename)
    semantics = RobotSemantics.from_srdf_file(srdf_filename, model)
    robot = RobotClass(model, semantics=semantics)
    robot = client.load_robot(load_geometry=load_geometry, local_cache_directory=local_directory)
    robot.artist = RobotArtist(robot.model)
    robot.scale(1000)
    if client:
        robot.client = client
    return robot

with RosClient() as client:
    robot = Robot(client,"fanuc_m6ib")
    frames = []
    frames.append(Frame([0.0, 0.1, 0], [1, 0, 0], [0, 1, 0]))
    frames.append(Frame([0.5, 0.1, 0.6], [1, 0, 0], [0, 1, 0]))
    start_configuration = Configuration.from_revolute_values([-1.9, -1.6, -2, 0.4,0.6,0.9])
    jc = JointConstraint("joint_3", 0, 0.1)
    trajectory = robot.plan_cartesian_motion(frames,
                                             start_configuration,
                                             max_step=0.01,
                                             avoid_collisions=True,path_constraints=[jc])
    robot.inverse_kinematics(frames[0], start_configuration, robot.main_group_name,constraints=[jc])
```
3. See error
```
File "<ipython-input-37-ca0940bb2c7f>", line 26, in <module>
    avoid_collisions=True,path_constraints=[jc])

  File "C:\ProgramData\Anaconda3\lib\site-packages\compas_fab\robots\robot.py", line 1257, in plan_cartesian_motion
    attached_collision_meshes=attached_collision_meshes)

  File "C:\ProgramData\Anaconda3\lib\site-packages\compas_fab\backends\ros\client.py", line 220, in plan_cartesian_motion
    return await_callback(self.plan_cartesian_motion_async, **kwargs)

  File "C:\ProgramData\Anaconda3\lib\site-packages\compas\utilities\async_.py", line 130, in await_callback
    async_func(*args, **kwargs)

  File "C:\ProgramData\Anaconda3\lib\site-packages\compas_fab\backends\ros\planner_backend_moveit.py", line 216, in plan_cartesian_motion_async
    self.GET_CARTESIAN_PATH(self, request, convert_to_trajectory, errback)

  File "C:\ProgramData\Anaconda3\lib\site-packages\compas_fab\backends\ros\planner_backend.py", line 100, in __call__
    return self.call(client, request, callback, errback)

  File "C:\ProgramData\Anaconda3\lib\site-packages\compas_fab\backends\ros\planner_backend.py", line 97, in call
    callback=inner_handler, errback=errback)

  File "C:\ProgramData\Anaconda3\lib\site-packages\roslibpy\core.py", line 262, in call
    self.ros.call_async_service(message, callback, errback)

  File "C:\ProgramData\Anaconda3\lib\site-packages\roslibpy\ros.py", line 248, in call_async_service
    self.factory.on_ready(_send_internal)

  File "C:\ProgramData\Anaconda3\lib\site-packages\roslibpy\comm\comm_autobahn.py", line 80, in on_ready
    callback(self._proto)

  File "C:\ProgramData\Anaconda3\lib\site-packages\roslibpy\ros.py", line 245, in _send_internal
    proto.send_ros_service_request(message, callback, errback)

  File "C:\ProgramData\Anaconda3\lib\site-packages\roslibpy\comm\comm.py", line 80, in send_ros_service_request
    json_message = json.dumps(dict(message)).encode('utf8')

  File "C:\ProgramData\Anaconda3\lib\json\__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)

  File "C:\ProgramData\Anaconda3\lib\json\encoder.py", line 199, in encode
    chunks = self.iterencode(o, _one_shot=True)

  File "C:\ProgramData\Anaconda3\lib\json\encoder.py", line 257, in iterencode
    return _iterencode(o, 0)

  File "C:\ProgramData\Anaconda3\lib\json\encoder.py", line 180, in default
    o.__class__.__name__)

TypeError: Object of type 'JointConstraint' is not JSON serializable
```

When I use similar code in Grasshopper , same error happened.

**Expected behavior**
> I want to add some joint or path constraints to make the result of inverse_kinematics more reliable, but when I add some contraints according documention ,the error happened  

**Desktop (please complete the following information):**
 - OS: win10
 - Python3.6 , pip

Please help me,thanks!

Qipccc, 2020-03-03:
when I use code like the follows ,it still the same  error:
```python
with RosClient() as client:    
    robot = Robot(client,"fanuc_m6ib")
    respose = robot.inverse_kinematics(Frame([11.4, 538, 810], [0,-1, 0], [-1, 0, 0]), start_configuration, robot.main_group_name)
    configuration = Configuration.from_revolute_values([1.3, 0, -0.7, -3.0, 1, 0.])
    tolerances = [math.radians(5)] * 6
    group = robot.main_group_name
    constraints_0 = robot.constraints_from_configuration(configuration, tolerances, group)
    respose2 = robot.inverse_kinematics(Frame([11.4, 538, 810], [0,-1, 0], [-1, 0, 0]), start_configuration, robot.main_group_name,constraints=constraints_0)
```

gonzalocasas, 2020-03-03:
@Qipccc you are right, the constraints are not being converted correctly. We just pushed a fix here: https://github.com/compas-dev/compas_fab/pull/113  Could you give it a try and confirm if it works for you? If so, we'll publish a patch release asap.

Qipccc, 2020-03-05:
Thank you for your replay. This method avoids the previous error, but when I try to use it, I find that it does not work, and a "NO_IK_SOLUTION" error will appear directly,  but actually it should have IK solution.

gonzalocasas, 2020-03-05:
I tested this with the panda model and it worked fine, so I guess something
in your setup is a bit different.
- Could it be scale is getting in the way?
- could you share a reproducible setup, including the robot model you use?

Qipccc, 2020-03-05:
Thank you for your prompt reply!Here this my gh file， I will appreciate it if you can test this.

 
 
 
 &gt; Thank you for your replay. This method avoids the previous error, but when
 &gt; I try to use it, I find that it does not work, and a "NO_IK_SOLUTION" error
 &gt; will appear directly, but actually it should have IK solution.
 &gt;
 &gt; —
 &gt; <https://github.com/compas-dev/compas_fab/issues/124?email_source=notifications&amp;email_token=AAHD3HJJKUVSNMHMSEM6CV3RF5KZNA5CNFSM4LAKIQP2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEN4CY5A#issuecomment-595078260&gt;,
 &gt; .
 &gt;

gonzalocasas, 2020-03-05:
@Qipccc could you please attach it via the Github web interface? Because it gets lost if you attach them to the reply email of github. Also, please attach the URDF models you are using.
Thanks!

gonzalocasas, 2020-03-05:
@Qipccc are you using the `fanuc_m6ib` models from the ROS repo https://github.com/ros-industrial/fanuc? If it's that, without local changes, I can just use that to verify/test.

Qipccc, 2020-03-05:
I am using the ur10 model from https://github.com/ros-industrial/universal_robot. 

[CompasConstraintTest.zip](https://github.com/compas-dev/compas_fab/files/4292281/CompasConstraintTest.zip)

gonzalocasas, 2020-03-05:
Aha! Ok.. I will test! But please check because you're passing `fanuc_m6ib` as the name, and it might be that it's using the cached version of that robot instead.

Qipccc, 2020-03-06:
@gonzalocasas  Actually ,I test both `fanuc_m6ib` and `ur10` ,they all looks like not worked, but I feel `ur10` looked better than `fanuc_m6ib` ,because I found *start_configuration* seems not worked in `fanuc_m6ib` when use *inverse_kinematic*

gonzalocasas, 2020-03-16:
@Qipccc I checked the UR10 model using your file and it actually works, but there are some notes.

If I use the values you set, were `shoulder_pan_joint` is constrained to `4.0` with a `1.5` tolerance, then I get very few solutions, say, 1 out of 5 tries it works.

For example, this one:
[image]

But it seems `4.0` is not a very good value for the selected plane, if I flip it to `-4.5`, then I get many more correct results:
[image]

So, I would say this works, but it is true that adding constraints makes it harder for the solver to find a solution and in some cases it gives up before finding one. If you wish to get a solution irrespective of the time, then use a higher value for the `attempts` parameter, e.g.:
```python
response = robot.inverse_kinematics(frame, start_configuration, group,constraints=[constraint], attempts=100)
```
using 100 attempts will almost always find a solution, at the expense of spending more time. The downside of using a very large value for `attempts` is that if you pass a plane that is really unreachable, Grasshopper will block for a long time and not be responsive and in the end, it will not find anything either.

gonzalocasas, 2020-03-16:
I will close this, as the base problem is resolved (`JSON serialization`) and merge the pull request. But please, feel free to keep discussing this topic here (even if closed) if you still have problems calculating IK.

closed 2020-03-16 by gonzalocasas