compas-dev/compas_fab #125: URDF loading problem

philianeles, 2020-03-05:
**Describe the bug**
Cannot create instance of <class 'compas.robots.model.geometry.Origin'>. Message=__init__() takes at most 4 arguments (5 given)

**To Reproduce**
Steps to reproduce the behavior:
1. Rhino 6, Grasshopper.
2. See the URDF below.
3. See the URDF below.
4. See error:
```
Runtime error (TypeErrorException): Cannot create instance of <class 'compas.robots.model.geometry.Origin'>. Message=__init__() takes at most 4 arguments (5 given)

Traceback:
  line 143, in parse_element, "C:\Users\sercan\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\files\urdf.py"
  line 128, in parse_element, "C:\Users\sercan\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\files\urdf.py"
  line 128, in parse_element, "C:\Users\sercan\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\files\urdf.py"
  line 48, in __init__, "C:\Users\sercan\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\files\urdf.py"
  line 65, in from_file, "C:\Users\sercan\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\files\urdf.py"
  line 103, in from_urdf_file, "C:\Users\sercan\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas\robots\model\robot.py"
  line 30, in script
```

**Expected behavior**
> Robot model loaded successfully.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Windows 10
 - Python 3.7
 - conda

**Additional context**
- URDF here: https://github.com/gramaziokohler/rob_tech_on_site_construction/blob/master/docs/robot_description/chiron/robot_description.urdf

romanarust, 2020-03-05:
Hi @philianeles! in line  283 of your file you have 4 values for origin, but only 3 are allowed (x y z): 
`<origin rpy="-1.57079632679 -1.57079632679 0" xyz="0 0.3124 -0.0004177 0"/>`
That's the problem.

romanarust, 2020-03-05:
Closing this, it should work when you change it.

closed 2020-03-05 by romanarust

philianeles, 2020-03-05:
👌