compas-dev/compas_fab #126: Whitespace issues in Sphinx html Example/Doctest

tetov, 2020-03-09:
## Description

Multi line statements in doctests with `"\\n"` are nicely legible in source code but renders without linebreaks and with tabs as spaces in Sphinx html output.

### Example from [`compas_fab.robots.Robot.plan_cartesian_motion`](https://gramaziokohler.github.io/compas_fab/latest/reference/generated/compas_fab.robots.Robot.plan_cartesian_motion.html)

#### Input 

[Source](https://github.com/compas-dev/compas_fab/blob/097dd726d7e59da7cf2e0d93683ee722479db0b3/src/compas_fab/robots/robot.py#L1216)
```python
        """
        Examples
        --------
        >>> frames = [Frame([0.3, 0.1, 0.5], [1, 0, 0], [0, 1, 0]),\
                      Frame([0.4, 0.3, 0.4], [0, 1, 0], [0, 0, 1])]
        >>> start_configuration = Configuration.from_revolute_values([-0.042, 4.295, -4.110, -3.327, 4.755, 0.])
        >>> group = robot.main_group_name
        >>> trajectory = robot.plan_cartesian_motion(frames,\
                                                     start_configuration,\
                                                     max_step=0.01,\
                                                     jump_threshold=1.57,\
                                                     avoid_collisions=True,\
                                                     group=group)
        >>> type(trajectory)
        <class 'compas_fab.robots.trajectory.JointTrajectory'>
        """
```

#### Output

[image]

## To Reproduce

1. Write doctest with escaped line breaks.
2. `Invoke docs`

## Expected behavior**
Can we find a way to make sphinx render the line breaks properly?

gonzalocasas, 2020-03-09:
well.. I never found any way.. that's why the examples were in one gigantic line before. ;)

tetov, 2020-03-09:
Ok, I'll look into it a bit more. We can't be the only ones with this issue.

Close?

gonzalocasas, 2020-03-09:
yup

closed 2020-03-09 by gonzalocasas