compas-dev/compas_fab #128: Bump dependency to COMPAS 0.15.x

gonzalocasas, 2020-03-16:
Bump dependency to the latest release of COMPAS (`0.15.4` as of today), which brings several breaking changes, especially to the `Mesh` datastructure.

philianeles, 2020-04-07:
I want to upgrade to COMPAS `0.15.5` - I need a method that is released in that version - what would you recommend? (compas_fab version)?

yck011522, 2020-04-20:
Hi @gonzalocasas .

Is this bump to compas 0.15.x  going to happen soon?

yck011522, 2020-04-20:
Okay, after a few more hours of struggling. I decided I cannot really work with COMPAS 0.13.x because there is no more online documentation online for this old version. (Not going to bother compiling my own old doc) And the breaking change of Mesh means I better develop with the latest COMPAS so I don't have to change my code later.

So I'm going to stop using compas_fab for now and bump 💥 this issue from time to time until this gets implemented. Can't access the lab until COVID-19 is fixed anyways.

gonzalocasas, 2020-04-30:
https://github.com/compas-dev/compas_fab/pull/152

gonzalocasas, 2020-05-07:
COMPAS v0.11.0 is out on conda & pip

closed 2020-05-07 by gonzalocasas

yck011522, 2020-05-07:
💥 Yey. This is amazing.