compas-dev/compas_fab #142: Scale method using compas.geometry.Scale or scale factor

tetov, 2020-04-15:
Classes found in `compas_fab.robots` often have a `scale` method. When it comes to joints, links and constraints of those the scaling is only a multiplication. When it comes to geometry, as in `BoundingVolume` and `CollisionMesh` (examples below), there's two different methods. 

`BoundingVolume`'s method is a bit limiting since it only allows for uniform scaling, but that could be changed by allowing a list of scale factors.

`CollisionMesh` accepts any `compas.geometry.Transformation`. But maybe the `scale` method should only scale the geometry. We can always add a `transform` method, [like in `BoundingVolume`](https://github.com/compas-dev/compas_fab/blob/788739f2d1f34f8975c848f37514df4aa5270334/src/compas_fab/robots/constraints.py#L97).

from [`compas_fab.robots.BoundingVolume`](https://github.com/compas-dev/compas_fab/blob/788739f2d1f34f8975c848f37514df4aa5270334/src/compas_fab/robots/constraints.py#L93)
```python
 def scale(self, scale_factor):
        S = Scale([scale_factor] * 3)
        self.transform(S)
```
from [`compas_fab.robots.CollisionMesh`](https://github.com/compas-dev/compas_fab/blob/788739f2d1f34f8975c848f37514df4aa5270334/src/compas_fab/robots/planning_scene.py#L46)
```python
def scale(self, transformation):
        """Scales the collision mesh.
        Parameters
        ----------
        transformation : compas.geometry.Scale
        """
        mesh_transform(self.mesh, transformation)
```

I can open a PR with these changes if this makes sense :).

romanarust, 2020-04-16:
Thanks for pointing out the inconsistency!
In that context, `scale` is always used uniformely and actually only used to convert robot units from m to cm, or mm, depending on the preferred planning units. So, scaling by only one factor is sufficient. If you could change the method `CollisionMesh.scale` to take only a scale_factor, that would be great.

But maybe it would be good here to reconsider the `scale` at all:
* should't `scale` be rather `scale_by_factor`?
* or even `scale_to_mm`, `scale_to_cm`, `scale_to_m`?
@gonzalocasas: any thoughts?

tetov, 2020-04-17:
Fixed in #144

closed 2020-04-17 by tetov

matteo-pacher, 2020-04-21:
> But maybe it would be good here to reconsider the `scale` at all:
> 
> * should't `scale` be rather `scale_by_factor`?
> * or even `scale_to_mm`, `scale_to_cm`, `scale_to_m`?
>   @gonzalocasas: any thoughts?

I agree with Romana that `scale` in this context is used to handle different input units. Would it make sense to generally use an argument for the units (for example "m", "mm", "cm", ...), and a have function in the back that makes sure to transform from the input unit to the model unit (and basically determine the right scale factor)?
Should the robot class then also have a variable that stores the units (which I guess should be meters generally, but there could be exceptions for models that we did not create ourselves)?