compas-dev/compas_fab #146: Missing description for inherited class attributes in docs

tetov, 2020-04-17:
Class attributes (like `compas_fab.robots.Constraints.JOINT`) show up as an attribute if documented in source code both in parent class and child classes. However, in child classes they show up without description. Compare docs from pr #138: Parent [`Constraint`](https://p.tetov.se/robots_constraints/reference/generated/compas_fab.robots.Constraint.html) & child [`JointConstraint`](https://p.tetov.se/robots_constraints/reference/generated/compas_fab.robots.JointConstraint.html).

I see three ways to fix this:
* Figure out why their description is lost for the child. (But picking up class attributes is not well documented, so might be tricky)
* Skip showing inherited attributes for child classes. 
* Make a distinction between inherited and non-inherited attributes (and methods) using a tweak to the `docs/_templates/autosummary/class.rst` like [this](https://github.com/compas-dev/compas/blob/master/docs/_templates/autosummary/class.rst.copy). That would guide the user to the parent class for more information about the class attributes.

tetov, 2020-04-17:
https://github.com/sphinx-doc/sphinx/issues/741

gonzalocasas, 2021-02-23:
@tetov this looks like fixed, right?

tetov, 2021-02-23:
Yeah, fixed in sphinx `3.5`, and looks good on the website. CI used `3.5.1` in latest run, but I'd prefer explicitly specifying `sphinx>=3.5` in `requirements-dev.txt`. What do you think?

gonzalocasas, 2021-02-23:
> but I'd prefer explicitly specifying sphinx>=3.5 in requirements-dev.txt. What do you think?

agree!

I'll close this for now, since the main issue is resolved, but +1 to bump the dep

closed 2021-02-23 by gonzalocasas