compas-dev/compas_fab #148: Retrieve all COMPAS meshes of a robot model

gonzalocasas, 2020-04-23:
# Feature Request

(Outcome of a conversation with @yck011522)

There is currently not an easy way to get all the meshes (COMPAS ones, not native CAD ones) of a robot model. The options are to either write a very verbose function that is close to `BaseRobotArtist.create()`, or to write a fake artists that inherits from `BaseRobotArtist` and simply returns the COMPAS meshes instead of converting to native.

## Details

This might be implemented directly in the `BaseRobotArtist` as the default implementation, or we could add an artist that does this, but I would have trouble finding a fitting name for such an artist. In any case, here's one tiny hack of an implementation of such a thing:

```python
class FakeRobotArtist(BaseRobotArtist):
    def transform(self, native_mesh, transformation):
        compas_mesh.transform(transformation)
    def draw_geometry(self, geometry, name=None, color=None):
        return mesh.copy()  # I think a copy would be required, to avoid screwing up stuff from the source model
```

gonzalocasas, 2021-08-18:
https://github.com/compas-dev/compas/pull/882

closed 2021-08-30 by beverlylytle

yck011522, 2021-08-31:
Yay