compas-dev/compas_fab #155: robot.zero_configuration()

matteo-pacher, 2020-05-20:
**Describe the bug**
The `robot.robot.zero_configuration()` function assigns `0` to every joint value in the configuration of the robot. This can be an invalid configuration for some robots, such as the RFL model, where the 0 values for many joints is out of limits. This creates an invalid start configuration (invalid bounds) for planning services calls.

**Expected behavior**
MoveIt and Rviz face the same issue when loading the robot. As described in http://wiki.ros.org/joint_state_publisher, _"The default value is zero. If zero is not a permissible value (max+min)/2 is used"_. I wonder how to detect permissible values.

**Screenshots**
[image]

xarthurx, 2020-05-20:
Not sure how formally this should be done, but based on my experience, use sth like "0.0001" will solve issues like this sometimes.

closed 2020-05-20 by romanarust

yck011522, 2020-05-20:
I think one alternative solution is to add an intrinsic property to the RobotModel to save the home position.

Many CNC controllers have such properties such as park position, home position, to remember a fixed position to go back to. And I believe this is a similar case i for robots.

matteo-pacher, 2020-05-20:
Thank you Romana!
For the RFL moveit_config, there is a defined home configuration for each planning group, which I guess could be retrieved. But that would be a backend dependent implementation.

romanarust, 2020-05-20:
Well, yes, it is stored in the srdf file. Feature request?

matteo-pacher, 2020-05-20:
it is stored in the srdf indeed, as group_state:
```
<!--GROUP STATES: Purpose: Define a named state for a particular group, in terms of joint values. This is useful to define states like 'folded arms'-->
    <group_state name="robot11_home" group="robot11">
        <joint name="robot11_joint_1" value="0" />
        <joint name="robot11_joint_2" value="0" />
        <joint name="robot11_joint_3" value="0" />
        <joint name="robot11_joint_4" value="0" />
        <joint name="robot11_joint_5" value="0" />
        <joint name="robot11_joint_6" value="0" />
    </group_state>
    <group_state name="robot12_home" group="robot12">
        <joint name="robot12_joint_1" value="0" />
        <joint name="robot12_joint_2" value="0" />
        <joint name="robot12_joint_3" value="0" />
        <joint name="robot12_joint_4" value="0" />
        <joint name="robot12_joint_5" value="0" />
        <joint name="robot12_joint_6" value="0" />
    </group_state>
    <group_state name="robot11_eaXYZ_home" group="robot11_eaXYZ">
        <joint name="bridge1_joint_EA_X" value="0" />
        <joint name="robot11_joint_1" value="0" />
        <joint name="robot11_joint_2" value="0" />
        <joint name="robot11_joint_3" value="0" />
        <joint name="robot11_joint_4" value="0" />
        <joint name="robot11_joint_5" value="0" />
        <joint name="robot11_joint_6" value="0" />
        <joint name="robot11_joint_EA_Y" value="0" />
        <joint name="robot11_joint_EA_Z" value="-4.915" />
    </group_state>
    <group_state name="robot12_eaYZ_home" group="robot12_eaYZ">
        <joint name="robot12_joint_1" value="0" />
        <joint name="robot12_joint_2" value="0" />
        <joint name="robot12_joint_3" value="0" />
        <joint name="robot12_joint_4" value="0" />
        <joint name="robot12_joint_5" value="0" />
        <joint name="robot12_joint_6" value="0" />
        <joint name="robot12_joint_EA_Y" value="-12.237" />
        <joint name="robot12_joint_EA_Z" value="-4.915" />
    </group_state>
    <group_state name="robot21_home" group="robot21">
        <joint name="robot21_joint_1" value="0" />
        <joint name="robot21_joint_2" value="0" />
        <joint name="robot21_joint_3" value="0" />
        <joint name="robot21_joint_4" value="0" />
        <joint name="robot21_joint_5" value="0" />
        <joint name="robot21_joint_6" value="0" />
    </group_state>
    <group_state name="robot22_home" group="robot22">
        <joint name="robot22_joint_1" value="0" />
        <joint name="robot22_joint_2" value="0" />
        <joint name="robot22_joint_3" value="0" />
        <joint name="robot22_joint_4" value="0" />
        <joint name="robot22_joint_5" value="0" />
        <joint name="robot22_joint_6" value="0" />
    </group_state>
    <group_state name="robot21_eaXYZ_home" group="robot21_eaXYZ">
        <joint name="bridge2_joint_EA_X" value="39.805" />
        <joint name="robot21_joint_1" value="0" />
        <joint name="robot21_joint_2" value="0" />
        <joint name="robot21_joint_3" value="0" />
        <joint name="robot21_joint_4" value="0" />
        <joint name="robot21_joint_5" value="0" />
        <joint name="robot21_joint_6" value="0" />
        <joint name="robot21_joint_EA_Y" value="0" />
        <joint name="robot21_joint_EA_Z" value="-4.915" />
    </group_state>
    <group_state name="robot22_eaYZ_home" group="robot22_eaYZ">
        <joint name="robot22_joint_1" value="0" />
        <joint name="robot22_joint_2" value="0" />
        <joint name="robot22_joint_3" value="0" />
        <joint name="robot22_joint_4" value="0" />
        <joint name="robot22_joint_5" value="0" />
        <joint name="robot22_joint_6" value="0" />
        <joint name="robot22_joint_EA_Y" value="-12.237" />
        <joint name="robot22_joint_EA_Z" value="-4.915" />
    </group_state>
```

romanarust, 2020-05-20:
#156

commit 2a1a83fa (2020-05-20): fixes #155