compas-dev/compas_fab #160: Robot's get_link_names throws exception for group with one link

beverlylytle, 2020-06-04:
**Describe the bug**
Robot's get_link_names throws exception for group with one link

**To Reproduce**
Steps to reproduce the behavior:
1. Load the ur5 demo robot via ROS.
2. Run the following code:
`
from compas_fab.backends import RosClient
from compas_fab.robots.ur5 import Robot

with RosClient() as client:
    robot = Robot(client)
    names = robot.get_link_names('endeffector')
`
**Expected behavior**
`names` should equal `['ee_link']`

**Actual behavior**
Exception is thrown "Exception: No chain found between the specified element"

**Desktop (please complete the following information):**
 - OS: OSX
 - Python version 3.8
 - Python package manager conda

kaniket7209, 2022-01-14:
Hey @gonzalocasas I want to start my contribution in this good first issue .

utam-1, 2024-06-16:
Hi @gonzalocasas, will you assign this issue to me? I have just began learning about open source.

gonzalocasas, 2024-06-16:
> Hi @gonzalocasas, will you assign this issue to me? I have just began learning about open source.

Done! Thanks!!!

gonzalocasas, 2024-06-16:
I will update the example code to the latest version because the one on the description is very old

gonzalocasas, 2024-06-16:
I've updated the snippet of code that will cause this bug to make it easier to reproduce. If you have problems while trying to fix this, feel free to post questions here!

annapurna-gupta, 2024-10-13:
Hi @gonzalocasas ,

I just submitted a PR that addresses an issue with the `get_link_names` method in the `compas_fab` library. This fix allows the method to handle groups containing only one link without throwing exceptions.

Could you please take a look when you have a moment? Your feedback would be greatly appreciated!

Thank you!

gonzalocasas, 2025-06-03:
(btw, looking at the description I understand why you wrote the test like that, but that was just an example piece of code, not how a unit test is supposed to be written)