compas-dev/compas_fab #166: Add examples for scaling a robot.

yck011522, 2020-06-04:
# Feature Request

It would be nice to have an example showing how to load a robot from URDF / ROS (robot in meters) and then scale it (for millimeters).

```
robot = client.load_robot(load_geometry=True)
robot.artist = RobotArtist(robot.model)
robot.scale(1000)
```

It is not clear to me whether scale() should be before setting artist or vise versa. In my opinion, both order should work but currently only my example will work.

It is also not clear to me how the scaling affects path planning input and output. Are those values also scaled or are they in native robot units?

It is also not clear to be whether scale() is a modal setting (setting a parameter) or an action (like a transformation). For example, the behavior of repeatedly calling robot.scale is not documented. I personally think it should behave as a modal setting to avoid potential issues.

pblcc, 2021-02-25:
Hello good! I would like to add this example to the repository in a pr. Do you have a **directory or folder where you would prefer me to add the example**? or just in the root of the directory?

gonzalocasas, 2021-02-26:
Hi @pblcc ! Great! Yes, we have a place that would make sense, I would suggest to put these examples under the `examples/02_description_models` together with other related examples: https://github.com/compas-dev/compas_fab/tree/main/docs/examples/02_description_models

Thanks!

karshav16, 2024-10-04:
Hello would love to contribute !

yck011522, 2024-10-05:
Hello @karshav16 thank you for the initiative. However, we are planning for a new major release where the scaling mechanism will change substantially. I think lets not work on this issue soon.