compas-dev/compas_fab #170: AttachedCollisionMesh in robot.inverse_kinematics()

matteo-pacher, 2020-06-05:
**Describe the bug**
`AttachedCollisionMesh` does not constitute a collision object when calling `robot.inverse_kinematics()`, even if `avoid_collisions=True`.

**To Reproduce**
```
from compas.geometry import Frame
from compas_fab.backends import RosClient
from compas.datastructures import Mesh
from compas_fab.robots import CollisionMesh, AttachedCollisionMesh

group = "robot11"

frame = Frame([1, 1, 1], [1, 0, 0], [0, -1, 0])

mesh_vertices = [[-0.5, -0.5, -0.5], [0.5, -0.5, -0.5], [0.5, -0.5, 0.5],
                 [-0.5, -0.5, 0.5], [0.5, -0.5, -0.5], [0.5, 0.5, -0.5],
                 [0.5, 0.5, 0.5], [0.5, -0.5, 0.5], [0.5, 0.5, -0.5],
                 [-0.5, 0.5, -0.5], [-0.5, 0.5, 0.5], [0.5, 0.5, 0.5],
                 [-0.5, 0.5, -0.5], [-0.5, -0.5, -0.5], [-0.5, -0.5, 0.5],
                 [-0.5, 0.5, 0.5], [-0.5, -0.5, -0.5], [-0.5, 0.5, -0.5],
                 [0.5, 0.5, -0.5], [0.5, -0.5, -0.5], [-0.5, -0.5, 0.5],
                 [0.5, -0.5, 0.5], [0.5, 0.5, 0.5], [-0.5, 0.5, 0.5]]
mesh_faces = [[0, 1, 2, 3], [4, 5, 6, 7], [8, 9, 10, 11], [12, 13, 14, 15], [16, 17, 18, 19], [20, 21, 22, 23]]
mesh = Mesh.from_vertices_and_faces(mesh_vertices, mesh_faces)

with RosClient('192.168.163.129') as client:

    robot = client.load_robot(load_geometry=False,
                              urdf_param_name='/robot_description',
                              srdf_param_name='/robot_description_semantic',
                              precision=None,
                              local_cache_directory=None)

    start_configuration = robot.zero_configuration()

    ee_link_name = robot.get_end_effector_link_name(group)
    acm = AttachedCollisionMesh(CollisionMesh(mesh, 'acm'), ee_link_name)

    ik_solution = robot.inverse_kinematics(frame,
                                           start_configuration=start_configuration,
                                           group=group,
                                           avoid_collisions=True,
                                           attached_collision_meshes=[acm])
                                           
    print(ik_solution)
```

**Expected behavior**
Like for attached collision meshes in `robot.plan_motion()`, the solver should theoretically throw an error.
robot.plan_motion() return such errors in the terminal:
```
[ INFO] [1591352868.935029847]: Found a contact between 'robot11_link_4' (type 'Robot link') and 'acm' (type 'Robot attached'), which constitutes a collision. Contact information is not stored.
[ INFO] [1591352868.935104793]: Collision checking is considered complete (collision was found and 0 contacts are stored)
[ INFO] [1591352868.935188688]: Start state appears to be in collision with respect to group robot11
[ WARN] [1591352868.979376739]: Unable to find a valid state nearby the start state (using jiggle fraction of 0.050000 and 100 sampling attempts). Passing the original planning request to the planner.
[ INFO] [1591352868.980154248]: Planner configuration 'robot11[RRT]' will use planner 'geometric::RRT'. Additional configuration parameters will be set when the planner is constructed.
[ WARN] [1591352868.980495615]: robot11[RRT]: Skipping invalid start state (invalid state)
[ERROR] [1591352868.980582843]: robot11[RRT]: There are no valid initial states!
[ INFO] [1591352868.980624983]: No solution found after 0.000250 seconds
[ WARN] [1591352868.991110936]: Goal sampling thread never did any work.
[ INFO] [1591352868.991488982]: Unable to solve the planning problem
```

**Screenshots**
RViz visualization of the attached collision mesh when added to the scene:
[image]

yck011522, 2020-06-12:
Hello @beverlylytle @gonzalocasas ,

I'm just wondering if there could be a hot-fix for this issue before the epic #149 is completed?

matteo-pacher, 2020-06-12:
in the _compas_fab\src\compas_fab\backends\ros\client.py_ file, look for the` inverse_kinematics(...)` definition, and add this line:
`kwargs['attached_collision_meshes'] = attached_collision_meshes`

beverlylytle, 2020-06-12:
I just pushed to master :shudder: exactly this.

yck011522, 2020-06-12:
Beautiful , you beautiful people.

beverlylytle, 2020-06-12:
Then I will close this issue, and attempt to remember to add this to the change log.

closed 2020-06-12 by beverlylytle