compas-dev/compas_fab #171: Bump dependency to compas 0.16

gonzalocasas, 2020-06-05:
Bump dependency to `compas 0.16`, which includes several changes to the transformation package, especially on the initialization of new transformation objects.

yck011522, 2020-06-05:
Yes, I just hit a brick today with this:

For example:
`CollisionMesh.scale()` does not work due to `S = Scale([scale_factor] * 3)` no longer valid.

tetov, 2020-06-06:
> Yes, I just hit a brick today with this:
> 
> 
> 
> For example:
> 
> `CollisionMesh.scale()` does not work due to `S = Scale([scale_factor] * 3)` no longer valid. 
> 
> 

I can fix this one, I recently worked on CollisionMesh scaling.

gonzalocasas, 2020-06-08:
#174

tetov, 2020-06-08:
I `grep`-ed around a bit in the repo and the only Transformation subclass instances created are Scale and Rotation. All Scale ones are changed in #174 and the Rotation ones are using an existing constructor.

tetov, 2020-06-08:
#174 
#177

gonzalocasas, 2020-06-15:
I believe this is completed already. Not aware of any remaining incompatibilities.

closed 2020-06-15 by gonzalocasas

philianeles, 2020-06-26:
I get this error while trying:
`conda install compas=0.16 compas_fab=0.11.0 `      
                                                                                                                                                                                                        UnsatisfiableError: The following specifications were found to be incompatible with each other:                                                                                                                                                                                                                                                                                                                                                                                                             Package compas conflicts for:                                                                                              
compas=0.16                                                                                                               
compas_fab=0.11.0 -> compas[version='>=0.15.6,<0.16']

tetov, 2020-06-28:
@philianeles 
> I get this error while trying:
> `conda install compas=0.16 compas_fab=0.11.0 `
> UnsatisfiableError: The following specifications were found to be incompatible with each other: Package compas conflicts for:
> compas=0.16
> compas_fab=0.11.0 -> compas[version='>=0.15.6,<0.16']

The bump to compas==0.16 has not been released yet, if you need to use compas==0.16 you could install the master branch.

```
conda install compas==0.16
pip install git+https://github.com/compas-dev/compas_fab
```