compas-dev/compas_fab #184: Neutral language refactoring

beverlylytle, 2020-07-06:
In line with the code of conduct, `compas_fab` should maintain current standards for neutral language.  In particular,`ros_master` should be renamed to `ros_core` wherever possible, the `master` branch should be renamed in accordance with GitHub's new standard, the promotional slides for `compas_fab` should be adapted, and any other non-neutral language in code or examples should be changed.

gonzalocasas, 2020-07-20:
ROS is changing to `main` as well: https://discourse.ros.org/t/consider-renaming-default-branches/15385/6

beverlylytle, 2020-07-21:
But from what I could tell in that discussion, it was only about git branch names, which I think effects us not at all.

gonzalocasas, 2020-07-21:
True, there was only a passing mention of the ros-master, but that is because ROS2 got rid of master/slave already. As far as I can tell, the removal was not related to neutral language, but more a side effect of the new `DDS` service used to coordinate nodes, but nevertheless, it did go away.

gonzalocasas, 2020-09-21:
Github's default branch will change to `main` starting 1st October, and apparently (as reported by [zdnet](https://www.zdnet.com/article/github-to-replace-master-with-main-starting-next-month/)), by the end of the year, they will have tools/features to make it seamless to update for existing repos.

beverlylytle, 2021-01-29:
This exists now: https://docs.github.com/en/github/administering-a-repository/renaming-a-branch

gonzalocasas, 2021-03-02:
Done! 
ROS-specific updates to be done on #272

closed 2021-03-02 by gonzalocasas