compas-dev/compas_fab #187: Controlling plan_motion key frame spacing (moveit)

yck011522, 2020-08-15:
Collision checking during `plan_motion` only happens at each point of the trajectory, sometimes small collision mesh features are not captured.
Is it possible to control how dense move-it generate these points such that I can force it to make a more accurate check?

romanarust, 2020-08-17:
This is possible by changing the MoveIt parameter: `maximum_waypoint_distance` or `longest_valid_segment_fraction`

```
import roslibpy
maximum_waypoint_distance = 0.3
param = roslibpy.Param(robot.client, '/move_group/ompl/maximum_waypoint_distance')
param.set(maximum_waypoint_distance, callback=print())
```

See here:
http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ompl_interface/ompl_interface_tutorial.html

yck011522, 2020-08-17:
@romanarust 

I have tried doing the following lines before my planning, but it doesn't seem to have much effect. There is no error but I cannot verify if OMPL actually took the settings.
```
import roslibpy
maximum_waypoint_distance = 0.000001
param = roslibpy.Param(robot.client, '/move_group/ompl/maximum_waypoint_distance')
param.set(maximum_waypoint_distance, callback=print())
longest_valid_segment_fraction = 0.001
param = roslibpy.Param(robot.client, '/move_group/ompl/longest_valid_segment_fraction')
param.set(longest_valid_segment_fraction, callback=print())
```
From what I see in the results, the waypoints are still sparse, `segment_fraction` is still having large jumps in between where small features are.

yck011522, 2020-08-17:
It is very hard to evaluate if the parameters are working correctly because the 'invalid' trajectory can only be checked manually. For example this the following is a planned motion that is "skipping" small features.

[image]

You can see that the last and second last waypoint here. The motion in between will collide with the scaffolding frame on the ground.
[image]

gonzalocasas, 2020-08-17:
> param.set(maximum_waypoint_distance, callback=print())

Don't use a callback there, a callback causes the param setting to be async, so it could happen that the value is not yet really set when you call the planner.

https://roslibpy.readthedocs.io/en/latest/reference/index.html#roslibpy.Param.set
[image]

yck011522, 2020-08-17:
@gonzalocasas . Thank for the tip. I can try again tomorrow but I don't think that's the case. I set a value and retry the planning for a number of times.

I believe this setting is persistent, so the next time I call planning. It should definitely have the value.

yck011522, 2020-08-19:
Update: I confirmed from ROS terminal with `rosparam get /move_group/ompl/maximum_waypoint_distance`  and the value is set properly. 

But the planned path do not have substantially more waypoints. And occasionally planned path is invalid as before.

gonzalocasas, 2020-08-19:
According to [this](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/ompl_interface/ompl_interface_tutorial.html), it seems like `longest_valid_segment_fraction` might be a better one; however, it does not seem to be dynamically reconfigurable (unless I'm reading the docs wrong), but it feels like `longest_valid_segment_fraction` needs to be changed in the `ompl_planning.yaml` file of your setup (I assume it's [this one](https://github.com/gramaziokohler/rfl_description/blob/master/rfl_moveit_config/config/ompl_planning.yaml)).

I would suggest updating them all in the yaml file (there are multiple instances if each, that apply to different planning groups), restart MoveIt and then retry.

BTW, this might be obvious, but just in case: these two params are in different units, one is a fraction of the space, and the other is absolute distance (in meters).