compas-dev/compas_fab #189: Is the `group` in `options` missing in `plan_cartesian_motion`

yck011522, 2020-08-18:
I see there is this new implementation of `plan_cartesian_motion` that has the `options` dictionary, And the following lines seems to be extract things from that `options` dict, is there any chance the `group` parameter is forgotten?

https://github.com/compas-dev/compas_fab/blob/f54d5bb4af975c523f8cffe9adaeab9a024c854b/src/compas_fab/robots/robot.py#L1210-L1220

beverlylytle, 2020-08-18:
`group` is left as an argument on the same level as `options`, not as a value within `options`.  It's passed in line 1218.

yck011522, 2020-08-18:
Hum.. yes you are right.... so maybe something else is off...

I'm trying to plan with one planning group (rfl's robot12_YZ) and the planner is giving me an error about robot11....

[image]

beverlylytle, 2020-08-18:
After discussion with @yck011522 , it was determined that the end effector for the main planning group was always being passed along in the request rather than the end effector for the specified group.

PR #190 (2020-08-18): get correct end effector
Closes #189 

### What type of change is this?

- [x] Bug fix in a **backwards-compatible** manner.
- [ ] New feature in a **backwards-compatible** manner.
- [ ] Breaking change: bug fix or new feature that involve incompatible API changes.
- [ ] Other (e.g. doc update, configuration, etc)
closed 2020-08-18 by beverlylytle