compas-dev/compas_fab #192: trouble scaling from meters to millimeters (+ other related issues when loading scene)

inesariza, 2020-08-20:
### **Describe the bug**
> Scaling of a robot from meters to millimeters does not seem to work properly
> There are other aspects of the setup not loading properly that may be related

### **To Reproduce**
grasshopper file including geometry inputs in meters and millimeters

**when in m:**
* robot loads fine from urdfs or from ROS
* collision mesh can be loaded only if loading robot directly from urdf

**when in mm:**
* robot loads fine from ROS if switching unit in gh file, but setup is different from meters setup

mm:
[image]
m:
[image]

* robot fails to load directly from the urdfs
* collision mesh cannot be loaded:
```Runtime error (PythonException): This method is only callable once a client is assigned

Traceback:
  line 7, in script
  line 111, in ensure_client, "C:\Users\Ines\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas_fab\robots\planning_scene.py"
  line 139, in add_collision_mesh, "C:\Users\Ines\AppData\Roaming\McNeel\Rhinoceros\6.0\Plug-ins\IronPython (814d908a-e25c-493d-97e9-ee3861957f49)\settings\lib\compas_fab\robots\planning_scene.py"
```
 * calculated path is not located where the geometry input is located, it is sort of moved the distance from the urdf model location:
[image]

### **Expected behavior**
* load from ROS in mm (working but with a different setup) + be able to load collision meshes (not working in mm)
* load directly from an urdf in mm

I am not sure I understand the implications of loading from ROS or urdfs, what is more practical, etc.

thank you!!

**Desktop (please complete the following information):**
 - OS: `Windows 10`
 - Python version `3.7`
 - Python package manager `conda`

gonzalocasas, 2020-08-20:
Which version of `compas_fab` are you using?
There are several unrelated issues here, but I think one big difference is that when you load from urdf (not from ROS), you don't have any client attached, and without a client, it is not possible to attach meshes. (That's why the error in script line 7 fails on `ensure_client`)

inesariza, 2020-08-20:
compas_fab 0.11.0

ok, good point. but:
 when working in meters the collision attaches correctly when loading from the urdf 😕

[waam_setup_m_mm.zip](https://github.com/compas-dev/compas_fab/files/5102949/waam_setup_m_mm.zip)