compas-dev/compas_fab #210: Cannot import helpers from compas_rhino

philianeles, 2020-09-21:
**Describe the bug**
cannot import helpers from compas_rhino

**To Reproduce**
Steps to reproduce the behavior:
1. Context: Rhino 6, Grasshopper
2. Sample script: see screen shot
3. Sample data: -
4. See error: see screen shot

**Expected behavior**
import helpers from compas_rhino

**Screenshots**
[image]

**Desktop:**
 - OS: Windows 10
 - Python version 3.7
 - Python package manager conda

gonzalocasas, 2020-09-21:
@philianeles yes, this is definitely a version issue (`compas_rhino.helpers` is gone in the 0.16.2 release). The latest `compas_fab` from the main branch runs correctly against `compas 0.16.2`.

philianeles, 2020-09-21:
@gonzalocasas thanks! it works.

closed 2020-09-21 by philianeles