compas-dev/compas_fab #214: robot-playground.ghx not working

matteo-pacher, 2020-09-24:
Two problems related to this file:
1) in the `compas_fab/docs/examples/03_backends_ros/06_ros_in_grasshopper.rst` doc, the link to download the `files/robot-playground.ghx` seems to be broken. Or it just does not work for me. The `robot-playground.ghx` file is actually present in the `files` subfolder though, and downloadble, but....
2) I cannot open the ghx file in Grasshopper, since it throws an error: "IO generated 1 messages, would you like to see them?"
[image]

beverlylytle, 2020-09-24:
I'm not able to reproduce this... The link works for me and I can load the file.  I will say that recently I had some troubles with this because of an improperly installed compas_rhino.  Uninstalling and reinstalling (with the version of install that exists in the latest version of compas) fixed this.

gonzalocasas, 2020-11-16:
@matteo-pacher does it work for you now? could we close this?

matteo-pacher, 2020-11-16:
all seems to work now 👍

closed 2020-11-16 by matteo-pacher