compas-dev/compas_fab #219: Robot description not loading correctly

philianeles, 2020-10-07:
**Describe the bug**
Recently upgraded to version 0.11.0, robot model is not loading correctly (axes shifted), had no problems with the same urdf with version 0.10.2

**To Reproduce**
Steps to reproduce the behavior:
1.Rhino 6
2. Sample code:
```
from scriptcontext import sticky as st
import os
import compas
import compas_fab
from compas.robots import RobotModel
from compas.robots import LocalPackageMeshLoader
from compas_fab.ghpython import RobotArtist
from compas_fab.robots import Robot as Robot

compas.PRECISION = '12f'

# store data in component
guid = str(ghenv.Component.InstanceGuid)
robot_key = "robot_" + guid

if robot_key not in st:
    st[robot_key] = None

if load_urdf == True:
    model = RobotModel.from_urdf_file(os.path.join(urdf_source, 'robot_description.urdf'))
    loader = LocalPackageMeshLoader(urdf_source, 'robot_description')
    model.load_geometry(loader)
        
    artist = RobotArtist(model)
    
    robot = Robot(model, artist, semantics=None, client=None)

    st[robot_key] = robot

```
3. urdf here:
```
<?xml version="1.0" encoding="utf-8"?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from /home/waco/catkin_ws/src/mabi_common/mabi_description/urdf_src/mabi_standard.urdf.xacro | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<!--Copyright (C) 2015 by Dario Bellicoso                                    -->
<!--bellicoso@mavt.ethz.ch                                                   -->
<!--                                                                         -->
<!--This program is free software; you can redistribute it and/or modify     -->
<!--it under the terms of the Lesser GNU General Public License as published -->
<!--by the Free Software Foundation; either version 3 of the License, or     -->
<!--(at your option) any later version.                                      -->
<!--                                                                         -->
<!--This program is distributed in the hope that it will be useful,          -->
<!--but WITHOUT ANY WARRANTY; without even the implied warranty of           -->
<!--MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the             -->
<!--Lesser GNU General Public License for more details.                      -->
<!--                                                                         -->
<!--You should have received a copy of the Lesser GNU General Public License -->
<!--along with this program. If not, see <http://www.gnu.org/licenses/>.     -->
<!-- Description of the MABI 12 robotic arm. -->
<robot name="mabi_arm" xmlns:xacro="http://www.ros.org/wiki/xacro">
  <material name="black">
    <color rgba="0.0 0.0 0.0 1.0"/>
  </material>
  <material name="blue">
    <color rgba="0.0 0.0 0.8 1.0"/>
  </material>
  <material name="green">
    <color rgba="0.0 0.8 0.0 1.0"/>
[... 285 lines cut ...]
      <robotBaseLink>arm_mount</robotBaseLink>
      <statePublisherRate>100</statePublisherRate>
    </plugin>
  </gazebo>
  <gazebo>
    <plugin filename="librokubimini_gazebo_plugin.so" name="gazebo_force_torque_sensor">
      <robotNamespace>mabi</robotNamespace>
      <statePublisherRate>100</statePublisherRate>
    </plugin>
  </gazebo>
</robot>

```
4. See error below in screenshot

**Expected behavior**
Robot model is loaded correctly.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Python version 3.7
 -  conda

gonzalocasas, 2020-10-08:
@philianeles could you please attach the meshes (or the whole `mabi_description` package) as a zip file to this issue?

philianeles, 2020-10-08:
[mabi_mobile_smb.zip](https://github.com/compas-dev/compas_fab/files/5347625/mabi_mobile_smb.zip)
@gonzalocasas Yes sure! Attached...

philianeles, 2020-10-28:
@gonzalocasas would you have any hints what could be the problem? :)

beverlylytle, 2021-01-28:
Hi @philianeles !  I was looking into this today, and I discovered that one of the biggest differences between v0.10.2 and v0.11.0, at least as far as robot visualization is concerned, is that the earlier version doesn't take into account the data contained in the `origin` tags of a link's geometry while the newer version does.  If I visualize the urdf and stls that are included in you zip file using the latest versions of compas and compas_fab, the robot looks disconnected.  But if I change one line of code to ignore these `origin`s in the `visual` elements of the links, it looks connected again.  This leads me to believe that `robot_description.urdf` doesn't exactly match up with the meshes in the stls, that is, that the meshes were transformed so that their origins match up with their link's origin.  So I guess my suggestion would be to delete every `robot.link.visual.origin` element in `robot_description.urdf`.  (And maybe also `robot.link.collision.origin` if there are any.)

gonzalocasas, 2021-03-02:
@philianeles ping! ;)

beverlylytle, 2021-07-26:
Closing due to inactivity.  Please feel free to open another issue if you are still running into problems @philianeles.

closed 2021-07-26 by beverlylytle