compas-dev/compas_fab #231: Documentation for Rhino and GHPython artist classes are missing.

yck011522, 2020-10-22:
**Describe the bug**
Documentation for Rhino and GHPython artist classes are missing.
[image]
[image]

beverlylytle, 2020-10-22:
Yes, these are missing because the artists have been moved to compas.  The docs for the robot artists in compas probably should be looked at again, and a deprecation warning should be given here.

gonzalocasas, 2020-11-05:
Updated these pages with links to the new compas artists: https://gramaziokohler.github.io/compas_fab/latest/reference/compas_fab.rhino.html

closed 2020-11-05 by gonzalocasas