compas-dev/compas_fab #235: synchronous PlanningScene methods: freezing in IronPython

matteo-pacher, 2020-10-28:
`PlanningScene.add_collision_mesh` method freezes in IronPython (GH).

**To Reproduce**

Unit test:
```
from time import time

import compas
import compas_fab
from compas.datastructures import Mesh
from compas_fab.backends import RosClient
from compas_fab.robots import CollisionMesh, PlanningScene

mesh_vertices = [[-0.5, -0.5, -0.5], [0.5, -0.5, -0.5], [0.5, -0.5, 0.5],
                 [-0.5, -0.5, 0.5], [0.5, -0.5, -0.5], [0.5, 0.5, -0.5],
                 [0.5, 0.5, 0.5], [0.5, -0.5, 0.5], [0.5, 0.5, -0.5],
                 [-0.5, 0.5, -0.5], [-0.5, 0.5, 0.5], [0.5, 0.5, 0.5],
                 [-0.5, 0.5, -0.5], [-0.5, -0.5, -0.5], [-0.5, -0.5, 0.5],
                 [-0.5, 0.5, 0.5], [-0.5, -0.5, -0.5], [-0.5, 0.5, -0.5],
                 [0.5, 0.5, -0.5], [0.5, -0.5, -0.5], [-0.5, -0.5, 0.5],
                 [0.5, -0.5, 0.5], [0.5, 0.5, 0.5], [-0.5, 0.5, 0.5]]
mesh_faces = [[0, 1, 2, 3], [4, 5, 6, 7], [8, 9, 10, 11], [12, 13, 14, 15], [16, 17, 18, 19], [20, 21, 22, 23]]
static_collision_objects = Mesh.from_vertices_and_faces(mesh_vertices, mesh_faces)

print(compas.__version__)
print(compas_fab.__version__)

with RosClient('192.168.19.128') as client:

    robot = client.load_robot(load_geometry=False,
                              urdf_param_name='/robot_description',
                              srdf_param_name='/robot_description_semantic',
                              precision=None,
                              local_cache_directory=None)
    print("robot =", robot)

    scene = PlanningScene(robot)
    print(scene)

    start = time()

    collision_objects = CollisionMesh(static_collision_objects, "static_collision_objects")
    scene.add_collision_mesh(collision_objects, False)

    end = time()
    print("Elapsed time =",  (end - start), "seconds")
```

When executed in Python 3.7.9 from terminal, here is the output:
```
pybullet build time: Oct 16 2020 16:15:00
0.16.9
0.12.0-63649e84
robot = <compas_fab.robots.robot.Robot object at 0x0000021855C47508>
<compas_fab.robots.planning_scene.PlanningScene object at 0x0000021855AE3208>
Elapsed time = 0.43320202827453613 seconds
```

When executed in IronPython inside GH, the software freezes and never recovers.

**Desktop (please complete the following information):**
 - OS: `Windows 10`
 - Python version: `3.7.9` VS `IronPython`
 - Python package manager `conda`

closed 2020-11-04 by gonzalocasas

matteo-pacher, 2020-11-10:
Thank you @gonzalocasas! I re-tested it, and the issue is solved. 👍 

I also quickly compared performances in Python 3.7.9 VS IronPython with this script:
```
from time import time

import compas
import compas_fab
from compas.datastructures import Mesh
from compas_fab.backends import RosClient
from compas_fab.robots import CollisionMesh, PlanningScene

mesh_vertices = [[-0.5, -0.5, -0.5], [0.5, -0.5, -0.5], [0.5, -0.5, 0.5],
                 [-0.5, -0.5, 0.5], [0.5, -0.5, -0.5], [0.5, 0.5, -0.5],
                 [0.5, 0.5, 0.5], [0.5, -0.5, 0.5], [0.5, 0.5, -0.5],
                 [-0.5, 0.5, -0.5], [-0.5, 0.5, 0.5], [0.5, 0.5, 0.5],
                 [-0.5, 0.5, -0.5], [-0.5, -0.5, -0.5], [-0.5, -0.5, 0.5],
                 [-0.5, 0.5, 0.5], [-0.5, -0.5, -0.5], [-0.5, 0.5, -0.5],
                 [0.5, 0.5, -0.5], [0.5, -0.5, -0.5], [-0.5, -0.5, 0.5],
                 [0.5, -0.5, 0.5], [0.5, 0.5, 0.5], [-0.5, 0.5, 0.5]]
mesh_faces = [[0, 1, 2, 3], [4, 5, 6, 7], [8, 9, 10, 11], [12, 13, 14, 15], [16, 17, 18, 19], [20, 21, 22, 23]]
static_collision_objects = Mesh.from_vertices_and_faces(mesh_vertices, mesh_faces)

print(compas.__version__)
print(compas_fab.__version__)

with RosClient('192.168.19.128') as client:

    robot = client.load_robot(load_geometry=False,
                              urdf_param_name='/robot_description',
                              srdf_param_name='/robot_description_semantic',
                              precision=None,
                              local_cache_directory=None)
    print("robot =", robot)

    scene = PlanningScene(robot)
    print(scene)

    entire_start = time()

    for i in range(100):
        start = time()

        collision_objects = CollisionMesh(static_collision_objects, "static_collision_objects")
        scene.add_collision_mesh(collision_objects, False)

        end = time()
        print("cycle took",  (end - start), "seconds")
    
    entire_end = time()
    print("Entire for loop took",  (entire_end - entire_start), "seconds")
```

- **Python 3.7.9 output**:
```
(semiramis) C:\Users\pacherm\Documents\000_RFL_URDF>python 201028_PlanningScene.py
0.17.2
0.13.1-5beb2ae8
robot = <compas_fab.robots.robot.Robot object at 0x0000029598E9DA08>
<compas_fab.robots.planning_scene.PlanningScene object at 0x000002959873FB88>
cycle took 0.4609510898590088 seconds
cycle took 0.5063419342041016 seconds
cycle took 0.5054347515106201 seconds
cycle took 0.5051984786987305 seconds
cycle took 0.5052511692047119 seconds
cycle took 0.5147719383239746 seconds
cycle took 0.5070717334747314 seconds
cycle took 0.5074970722198486 seconds
cycle took 0.22136187553405762 seconds
cycle took 0.5060515403747559 seconds
[... 182 lines cut ...]
('cycle took', 0.01611328125, 'seconds')
('cycle took', 0.01287841796875, 'seconds')
('cycle took', 0.01288604736328125, 'seconds')
('cycle took', 0.011260986328125, 'seconds')
('cycle took', 0.01450347900390625, 'seconds')
('cycle took', 0.01287841796875, 'seconds')
('cycle took', 0.014495849609375, 'seconds')
('cycle took', 0.01450347900390625, 'seconds')
('cycle took', 0.01610565185546875, 'seconds')
('cycle took', 0.01448822021484375, 'seconds')
('cycle took', 0.01129150390625, 'seconds')
('cycle took', 0.01448822021484375, 'seconds')
('cycle took', 0.01288604736328125, 'seconds')
('cycle took', 0.0113067626953125, 'seconds')
('cycle took', 0.012847900390625, 'seconds')
('cycle took', 0.0128936767578125, 'seconds')
('cycle took', 0.01287841796875, 'seconds')
('cycle took', 0.01288604736328125, 'seconds')
('cycle took', 0.01610565185546875, 'seconds')
('Entire for loop took', 1.3763351440429688, 'seconds')
```

The performance is way different:
49.28 seconds for Python 3.7.9 VS 1.37 seconds for IronPython!
What is happening here?

gonzalocasas, 2020-11-10:
Your performance test is probably affected by the way Twisted library controls the event loop. Try adding the following lines at the very top of your scrips (before any other import) and give it another go:

```
from twisted.internet import reactor
reactor.timeout = lambda : 0.0001
```

commit 7676342e (2020-11-04): fix #235: sometimes IronPython screws up something and returns an old-style class for service response