compas-dev/compas_fab #236: [Pybullet backend] Attached collision object does not update after robot configuration is updated

yijiangh, 2020-10-29:
**Describe the bug**
As the title suggested. @beverlylytle 

**To Reproduce**
compas_fab version `0.12.0`:

MWE:
```python
import time

import compas_fab
from compas_fab.backends.pybullet import PyBulletClient
from compas_fab.robots import AttachedCollisionMesh
from compas_fab.robots import CollisionMesh
from compas_fab.robots import Configuration

from compas.datastructures import Mesh

with PyBulletClient() as client:
    urdf_filepath = compas_fab.get('universal_robot/ur_description/urdf/ur5.urdf')
    robot = client.load_robot(urdf_filepath)

    mesh = Mesh.from_stl(compas_fab.get('planning_scene/cone.stl'))
    cm = CollisionMesh(mesh, 'tip')
    acm = AttachedCollisionMesh(cm, 'ee_link')
    client.add_attached_collision_mesh(acm, {'mass': 1, 'robot': robot})
    input('Attached!')

    # update robot conf
    conf = Configuration.from_revolute_values([1.0,0,0, 0,0,0])
    client.set_robot_configuration(robot, conf)

    time.sleep(1)
    client.step_simulation()
    time.sleep(1)
    input('update conf!')

    client.remove_attached_collision_mesh('tip')
    input('Removed!')

    time.sleep(1)
```

At the `Attached` state:
[image]

At the `'update conf!` state:
[image]

At the `removed!` state:
[image]

**Expected behavior**
The attached collision object should stay attached to the robot's flange after the configuration update.

**Desktop (please complete the following information):**
 - OS: Windows
 - Python version: 3.7
 - Python package manager: pip

beverlylytle, 2020-10-29:
So it seems that there is some lag in the changes that are applied to the cone.  If you run the following, you can watch the cone catch up to the instantaneous movement of the robot, but it doesn't seem to end up in quite the right final position.  Which is very strange considering the constraint added in add_attached_collision_mesh is of type JOINT_FIXED.
```
import time

import compas_fab
from compas_fab.backends.pybullet import PyBulletClient
from compas_fab.robots import AttachedCollisionMesh, Configuration
from compas_fab.robots import CollisionMesh

from compas.datastructures import Mesh

with PyBulletClient() as client:
    urdf_filepath = compas_fab.get('universal_robot/ur_description/urdf/ur5.urdf')
    robot = client.load_robot(urdf_filepath)

    mesh = Mesh.from_stl(compas_fab.get('planning_scene/cone.stl'))
    cm = CollisionMesh(mesh, 'tip')
    acm = AttachedCollisionMesh(cm, 'ee_link')
    client.add_attached_collision_mesh(acm, {'mass': 1, 'robot': robot})

    conf = Configuration.from_revolute_values([1.0,0,0, 0,0,0])
    client.set_robot_configuration(robot, conf)

    count = 0
    while count < 30:
        time.sleep(.25)
        client.step_simulation()
        count += 1
    input('update conf!')

    client.remove_attached_collision_mesh('tip')
    input('Removed!')

yijiangh, 2020-10-29:
@beverlylytle Interesting. I think by add a `JOINT_FIXED` constraint, the pybullet adds an artificial force between the object and the robot's attached link and thus the kinematics chain of the robot is extended. After we `set_robot_configuration`, part of the robot's joints is updated and pybullet needs to integrate a few steps (in an ODE sense) to make the attached object catch up.

But I think from a user's perspective, this is not intuitive because if we are working on path planning, we expect that the attached object's pose is updated synchronously when the robot's configuration is updated and we can perform collision checking. If we always need to step the simulation a few steps whenever we do collision checking to get the attached object updated, this will significantly slow down the planning pipeline.

This is exactly why we had the `Attachment` class in `pybullet_planning`, and we had this [assign](https://github.com/yijiangh/pybullet_planning/blob/dev/src/pybullet_planning/interfaces/task_modeling/grasp.py#L24) function to trigger the attached object's pose update. And in `compas_fab_pychoreo`'s `PyChoreoClient`, we overwrite `compas_fab.backend.PyBulletClient`'s `add_attached_collision_mesh` function: https://github.com/yijiangh/compas_fab_pychoreo/blob/master/src/compas_fab_pychoreo/client.py#L97

yijiangh, 2020-10-29:
Also, I found that after the integration, the robot might be in a different configuration than the desired configuration.

In the example above, the desired configuration is:

[image]

And after `100` time steps (the integrator clearly converges - you will see that in the last few steps the robot is not moving anymore), the robot's configuration becomes:

[image]

beverlylytle, 2020-10-30:
Yeah, the mass of the cone is having an effect on the movement of the robot.  If you increase the mass to something absurdly high, you'll see the robot move and then be dragged back to the cone.

beverlylytle, 2021-02-22:
Closed by #259.

closed 2021-02-22 by beverlylytle