compas-dev/compas_fab #244: 'Configuration' object has no attribute 'keys'

philianeles, 2020-11-11:
**Describe the bug**
I have upgraded to 0.12.0 from 0.11.2 and I am having the below error with Configuration.

**To Reproduce**
Steps to reproduce the behavior:
1. Context: Rhino 6, Grasshopper
2. Sample script (see below)
3. Sample data
[image]

4. See error:
```
Runtime error (MissingMemberException): 'Configuration' object has no attribute 'keys'

Traceback:
  line 249, in _update, "C:\Users\sercan\AppData\Roaming\McNeel\Rhinoceros\6.0\scripts\compas\robots\base_artist\_artist.py"
  line 683, in compute_transformations, "C:\Users\sercan\AppData\Roaming\McNeel\Rhinoceros\6.0\scripts\compas\robots\model\robot.py"
  line 244, in update, "C:\Users\sercan\AppData\Roaming\McNeel\Rhinoceros\6.0\scripts\compas\robots\base_artist\_artist.py"
  line 134, in update_model, "<string>"
  line 148, in script
```

**Desktop (please complete the following information):**
 - OS: 10
 - Python version 3.7
 - Python package manager conda

gonzalocasas, 2020-11-11:
@philianeles i think you need to change the line 134 from `robot.artist.update` to the new `robot.update` and the error will go away.

gonzalocasas, 2020-11-11:
and you'll need to remove the `names` parameter as well (i think)

philianeles, 2020-11-11:
it worked with changing the `robot.artist.update` to `robot.update`, thank you @gonzalocasas!

closed 2020-11-11 by philianeles