compas-dev/compas_fab #278: Check for IK solution without breaking (ROS)

laurend, 2021-03-17:
Is there a way to check if an IK solution exists without breaking the script and throwing an error? 
If so, could an example of how to do this be added to the docs? 

For example, I would like to be able to first check if solution exists and then offer an alternative frame if it doesn't.

beverlylytle, 2021-03-17:
The usual way to handle exceptions in Python is with a `try`-`except` clause.  For example, if `frames` is a list of frames you'd like to check inverse kinematic solutions for in decreasing order of preference:
```
for frame in frames:
    try:
        ik_solution = client.inverse_kinematics(robot, frame)
        break
    except BackendError:
        print("trying again")
else:
    print("No ik solution found.")
```
Would something like that suit your purposes?

laurend, 2021-03-17:
Yes... `BackendError` works. Thanks!

beverlylytle, 2021-03-17:
Great!

closed 2021-03-17 by beverlylytle