compas-dev/compas_fab #302: PyBulletClient get_uid(robot)

yijiangh, 2021-05-19:
When I run the following script using `PybulletClient` at version 0.18.0:

```python
with PyBulletClient(connection_type="gui", verbose=False) as client:
    robot = client.load_robot(urdf_filename)
    robot_uid = client.get_uid(robot)
```

It doesn't work because `robot` does not have the attribute `attr`.

The following script does work though:

```python
with PyBulletClient(connection_type="gui", verbose=False) as client:
    robot = client.load_robot(urdf_filename)
    robot_uid = client.get_uid(client.get_cached_robot(robot))
```

This makes me wonder if this line and the lines followed should use `robot.attr` instead of `robot.attributes`?
https://github.com/compas-dev/compas_fab/blob/main/src/compas_fab/backends/pybullet/client.py#L187

@beverlylytle Is this intended behavior to have users always work with the cached robot? Thanks!

Also, would you mind confirming that `redirect_stdout` is working ([this line](https://github.com/compas-dev/compas_fab/blob/main/src/compas_fab/backends/pybullet/client.py#L203))?

When loading the robot with `verbose=False`, I still get the following printous:

```
argv[0]=
starting thread 0
started testThreads thread 0 with threadHandle 0000000000000170
argc=3
argv[0] = --unused
argv[1] =
argv[2] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
Version = 4.6.0 NVIDIA 452.41
Vendor = NVIDIA Corporation
Renderer = GeForce GTX 1650/PCIe/SSE2
b3Printf: Selected demo: Physics Server
starting thread 0
started MotionThreads thread 0 with threadHandle 0000000000000AF8
MotionThreadFunc thread started
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: robotA_tool0
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: robotA_base
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: robotA_waam_tool
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: No inertial data for link, using mass=1, localinertiadiagonal = 1,1,1, identity local inertial frame
b3Printf: b3Warning[examples/Importers/ImportURDFDemo/BulletUrdfImporter.cpp,126]:

b3Printf: robotA_waam_tcp
```

beverlylytle, 2021-05-26:
The pybullet assigned id is attached to the internal `cached_robot` and not an instance of `compas_fab.robots.Robot` or its `compas_fab.robots.Robot.model`.  The `cached_robot` is an instance of `compas.robots.RobotModel` that is based on the `compas_fab.robots.Robot.model` but can be modified by the `PyBullet` client, and when it is, the pybullet id changes.  So in a sense, yes, the user is working with the `cached_robot`, but we try to keep it behind the scenes as much as possible.

For me, the `redirect_stdout` is working.  There is an exception to when it is enabled even if `verbose=False` is given, and that's when code is being run with pytest.

yijiangh, 2021-05-26:
> The pybullet assigned id is attached to the internal cached_robot and not an instance of compas_fab.robots.Robot or its compas_fab.robots.Robot.model. 

Thanks for the explanation! So this is the canonical way to obtain the pybullet index of a given robot instance?  

```
robot_uid = client.get_uid(client.get_cached_robot(robot))
```

> For me, the redirect_stdout is working. There is an exception to when it is enabled even if verbose=False is given, and that's when code is being run with pytest.

Aha, I am exactly encountering this with pytest 😃

beverlylytle, 2021-05-26:
> Thanks for the explanation! So this is the canonical way to obtain the pybullet index of a given robot instance?

robot_uid = client.get_uid(client.get_cached_robot(robot))

Yep.

I tried for a while to make redirect_stdout and pytest play together nicely, but couldn't manage.

yijiangh, 2021-05-26:
Thanks! I guess not many users would need to access the pybullet body index, except for someone who is working on backend features 😄

closed 2021-05-26 by yijiangh