compas-dev/compas_fab #313: Error code: -18; INVALID_LINK_NAME

PingHsunTsai, 2021-07-05:
**Describe the bug**
Traceback:
  line 133, in await_callback, "C:\Users\PingHsunTsai\AppData\Roaming\McNeel\Rhinoceros\6.0\scripts\compas\utilities\azync.py"
  line 77, in forward_kinematics, "C:\Users\PingHsunTsai\AppData\Roaming\McNeel\Rhinoceros\6.0\scripts\compas_fab\backends\ros\backend_features\move_it_forward_kinematics.py"
  line 20, in __call__, "C:\Users\PingHsunTsai\AppData\Roaming\McNeel\Rhinoceros\6.0\scripts\compas_fab\backends\interfaces\backend_features.py"
  line 31, in forward_kinematics, "C:\Users\PingHsunTsai\AppData\Roaming\McNeel\Rhinoceros\6.0\scripts\compas_fab\backends\ros\planner.py"
  line 31, in forward_kinematics, "C:\Users\PingHsunTsai\AppData\Roaming\McNeel\Rhinoceros\6.0\scripts\compas_fab\backends\interfaces\client.py"
  line 1283, in forward_kinematics, "C:\Users\PingHsunTsai\AppData\Roaming\McNeel\Rhinoceros\6.0\scripts\compas_fab\robots\robot.py"
  line 29, in script

**To Reproduce**
Steps to reproduce the behavior:
1. Rhino 6, Grasshopper
2. See error:
[image]

**Expected behavior**
> call `forward_kinematics()` succesfully

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Python 3.7
 - Python package manager: conda

**Additional context**
Using the ur_robot_driver with real hardware (with move_it package)

gonzalocasas, 2021-07-05:
Could you please try the following:
1) print and paste in this issue the result of `robot.info()`
2) try adding this line after line 23: ```configuration.joint_names = robot.get_configurable_joint_names()```

Also, I'm assuming you have the latest `compas_fab` version, if that's not the case, please post the version info of compas and compas_fab.

philianeles, 2021-07-06:
Sure! 

1. print and paste in this issue the result of `robot.info()`
```
0. The robot's name is 'ur10'.
1. The planning groups are: ['endeffector', 'manipulator']
2. The main planning group is 'manipulator'.
3. The end-effector's name is 'ee_link'.
4. The robot has NO tool attached.
5. The base link's name is 'base_link'
6. The base_frame is: Frame(Point(0.000, 0.000, 0.000), Vector(1.000, 0.000, 0.000), Vector(0.000, 1.000, 0.000))
7. The robot's joints are:
8. 	* 'shoulder_pan_joint' is of type 'revolute' and has limits [6.283, -6.283]
9. 	* 'shoulder_lift_joint' is of type 'revolute' and has limits [6.283, -6.283]
10. 	* 'elbow_joint' is of type 'revolute' and has limits [6.283, -6.283]
11. 	* 'wrist_1_joint' is of type 'revolute' and has limits [6.283, -6.283]
12. 	* 'wrist_2_joint' is of type 'revolute' and has limits [6.283, -6.283]
13. 	* 'wrist_3_joint' is of type 'revolute' and has limits [6.283, -6.283]
14. The robot's links are:
15. ['base_link', 'shoulder_link', 'upper_arm_link', 'forearm_link', 'wrist_1_link', 'wrist_2_link', 'wrist_3_link', 'ee_link', 'base', 'tool0', 'world']
```

2. try adding this line after line 23: `configuration.joint_names = robot.get_configurable_joint_names()`
[image]

And the versions:
- compas_fab 0.17.2
- compas 0.13.1

gonzalocasas, 2021-07-07:
The call to FK tries to figure out which link you want to get fk for, which by default it should be `ee_link`, but maybe something in your setup is slightly different. Try to specify it explicitly:

```
robot.forward_kinematics(configuration, group, options=dict(link='ee_link'))
```

Also, just to make sure, `group` is set to `manipulator`, right?

philianeles, 2021-07-15:
@Gonzalo so after going back to the `calibration_devel` branch and running:
roslaunch ur_robot_driver ur10_bringup.launch limited:=true robot_ip:=192.168.0.10
roslaunch ur10_moveit_config ur10_moveit_planning_execution.launch limited:=true
roslaunch ur10_moveit_config moveit_rviz.launch config:=true

With this way of loading the robot model:
[image]

We have a different error in the `artist` (but the previous problem seems to be gone!)
But the good news is that we can execute a trajectory with the real hardware :)
if we can figure out this problem with `artist`, maybe it`s good to proceed like this (without 2 clients?)

beverlylytle, 2021-07-20:
As discussed over slack, this last error is eliminated when `load_geometry` is set to `True` in `load_robot`.

closed 2021-07-20 by beverlylytle