compas-dev/compas_fab #322: PyBullet client: check if collision meshes are convex

inesariza, 2021-08-24:
# Feature Request

As a user of the PyBullet client, I want to check if collision meshes are convex before any collision detection operation is performed.

## Details

**Is your feature request related to a problem? Please describe.**
Even if using this client often one could forget the implicit rule that meshes need to be convex. It would be great to check for this automatically.

**Describe the solution you'd like**
Ideally, when adding collision meshes to the scene, concave meshes raise a warning telling the user that a convex hull operation will be performed. The user can then decide if continuing with the convex hull or updating inputs (decomposing the input meshes to make them convex). 

**Describe alternatives you've considered**
Alternatively, the user gets a warning message for each convex hull operation including the name of the mesh that has been wrapped.

inesariza, 2021-08-30:
thank you for your assessment and reply, @beverlylytle! 

Sounds like the convexity check would be costly and would not really solve the problem (i.e., that we want convex meshes). 

Having the option of applying V-HACD would be best for most cases. For this solution then, the only limitation is that we need .objs instead of compas meshes?

beverlylytle, 2021-08-30:
Yes.  What could be done is adding a flag somewhere... maybe on `load_robot` and also the planning scene methods, which would apply V-HACD when the flag is True.  (Behind the scenes, the `PyBulletClient` already does a lot of converting compas meshes to obj files to load into the pybullet server.)  So, handling convexity would not be automatic, but semi-automatic ;)

PR #324 (2021-09-02): add concavity support in pybullet
Closes #322  

@inesariza is this helpful?  could you test it?  (running vhacd on every mesh in a robot is time consuming, so don't be surprised if loading a robot is slow.)

### What type of change is this?

- [ ] Bug fix in a **backwards-compatible** manner.
- [x] New feature in a **backwards-compatible** manner.
- [ ] Breaking change: bug fix or new feature that involve incompatible API changes.
- [ ] Other (e.g. doc update, configuration, etc)

### Checklist

_Put an `x` in the boxes that apply. You can also fill these out after creating the PR. If you're  [... cut]
closed 2021-09-02 by beverlylytle