compas-dev/compas_fab #327: ModuleNotFoundError: No module named 'compas.files.xml'

odellus, 2021-09-04:
**Describe the bug**
> The line `from compas_fab.backends import RosClient` results in the error `ModuleNotFoundError: No module named 'compas.files.xml'

**To Reproduce**
Steps to reproduce the behavior:
1. Context - `bash`
2. Steps to reproduce: `python3 -m pip install compas_fab & python3 -c 'from compas_fab.backends import RosClient'`
3. Sample data - N/A
4. Error message - `ModuleNotFoundError: No module named 'compas.files.xml'

**Expected behavior**
> I expect to be able to import RosClient from compas_fab.backends

**Screenshots**
> Add screenshots to help explain your problem (if applicable).

**Desktop (please complete the following information):**
 - OS: `Ubuntu 20.04 on Windows 10 WSL2`
 - Python version `3.8`
 - Python package manager `pip`

**Additional context**
> I'm running the ROS websocket through Docker though I doubt that matters. This seems like a Python issue.

gonzalocasas, 2021-09-04:
Thanks for the report @odellus! 

I just tested the `pip` install on Ubuntu and indeed, the documented one causes issues, I had other issues before the one you encountered, but they are most likely related. The recommended way to get `compas_fab` installed is via conda, but regardless, the `pip` install should definitely work.

Could you please try to install `cython` before installing any compas package? ie.  `python3 -m pip install cython --install-option="--no-cython-compile"` and only then `python3 -m pip install compas_fab`. I tried on a clean environment and that got rid of all install errors. Please let me know if that works and I will update the docs accordingly.

odellus, 2021-09-04:
```bash
python3 -m pip uninstall compas compas_fab
python3 -m pip install cython
python3 -m pip install compas_fab
```
worked. Thank you!
Closing.

closed 2021-09-04 by odellus