compas-dev/compas_fab #338: No persistent connection with Grasshopper component ROS Connect

tetov, 2022-01-25:
## Describe the bug

Grasshopper component "ROS Connect" does not provide a persistent connection. When the connect input is set to True momentarily (using a button component) a connection is made but the input changes to false (the button is de-pressed) the connection is closed.

If  I attach a Toggle component set to True, i.e. a persistent True is given as an input Rhino crashes.

I made a "normal" GhPython component using the similar code that works as expected, see repro.ghx. I'll put the component code at the bottom too.

I think this might be a limitation of the component workflow, but I'm not sure.

## To Reproduce

Steps to reproduce the behavior:
1. Rhino 7.13.21348.13001, 2021-12-14, compas_fab 0.22.0, Windows 10
2. Open grasshopper file in [repro.zip](https://github.com/compas-dev/compas_fab/files/7933205/repro.zip)

## Screenshots

[image]

## Expected behavior

Persistent connection after pressing button component.

## Desktop (please complete the following information):**
 - OS: 
     - Edition: Windows 10 Pro
     - Version: 21H2
     - OS build: 19044.1466
     - Experience: Windows Feature Experience Pack 120.2212.3920.0
 - Python version `3.9.9`
 - Python package manager: `conda`

## Component code

### Original

[compas_fab/src/compas_fab/ghpython/components/Cf_RosConnect/code.py](https://github.com/compas-dev/compas_fab/blob/v0.22.0/src/compas_fab/ghpython/components/Cf_RosConnect/code.py)

### Sample of working code

```python
from scriptcontext import sticky as st

from compas_fab.backends import RosClient
from compas_fab.ghpython.components import create_id

ip = ip or '127.0.0.1'
port = port or 9090

key = create_id(ghenv.Component, 'ros_client')
ros_client = st.get(key, None)

if connect:
    
    if ros_client:
        st[key].close()
        
    st[key] = RosClient(ip, port)
    st[key].run(5)

is_connected = ros_client.is_connected if ros_client else False
```

gonzalocasas, 2022-01-25:
I'm confused because the scenario you mention as crashing is the one I use all the time, and I just tested without issues:
[image]

Also works using your repro file:
[image]

So, I wonder what is different between your setup and mine. I have seen the connection crash from time to time, but usually it was due to other components down the line that were doing calculations on ROS without checking the connection was really active...

tetov, 2022-01-25:
Hm.. Could you please try with [the docker file from compas_rrc_start](https://github.com/compas-rrc/compas_rrc_start/blob/main/docker/virtual_controller/docker-compose.yml)?

I'll test with just roscore later..

closed 2022-01-25 by tetov