compas-dev/compas_fab #382: Plan_motion and Plan_cartesian_motion error when a tool is attached to robot

begums, 2023-04-28:
**Describe the bug**
Using the compas_fab grasshopper components, plan_motion and plan_cartesian_motion give the error below, if a tool is attached to the robot. When no tool is attached, the planning works and a trajectory is produced, even with attached collision meshes to the scene or the robot.

**To Reproduce**
Steps to reproduce the behavior:
1. Context:
Using compas_fab version 0.27.0 on Rhino 7, Grasshopper. 
Compas_fab components used: ROS Connect, Attached Tool, Plan Cartesian Motion, Plan Motion
2. See error:

```
Runtime error (PythonException): Message type moveit_msgs/GetCartesianPathRequest does not have a field start_state.attached_collision_objects.object.pose

Traceback:
  line 137, in await_callback, "C:\Users\saral\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas\utilities\azync.py"
  line 99, in plan_cartesian_motion, "C:\Users\saral\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas_fab\backends\ros\backend_features\move_it_plan_cartesian_motion.py"
  line 133, in __call__, "C:\Users\saral\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas_fab\backends\interfaces\backend_features.py"
  line 45, in plan_cartesian_motion, "C:\Users\saral\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas_fab\backends\ros\planner.py"
  line 35, in plan_cartesian_motion, "C:\Users\saral\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas_fab\backends\interfaces\client.py"
  line 1557, in plan_cartesian_motion, "C:\Users\saral\AppData\Roaming\McNeel\Rhinoceros\7.0\scripts\compas_fab\robots\robot.py"
  line 24, in RunScript, "<string>"
```

**Expected behavior**
Expect it to produce a trajectory with a tool attached as well.

**Screenshots**
[image]

**Desktop**
 - OS: `[Windows 10]`
 - Python version `[3.10.8]`
 - Python package manager `[conda]`

gonzalocasas, 2023-04-28:
Thanks for the report! Quick question: which Ros version are you using?

begums, 2023-05-02:
@gonzalocasas Hey Gonzalo! I've tried it both with the noetic and melodic version. I used this docker image [https://hub.docker.com/r/gramaziokohler/ros-noetic-moveit](url) for the noetic.

gonzalocasas, 2023-05-02:
That's odd. Would you mind sharing / attaching the GH file and also the docker compose file you are using? Thanks!

begums, 2023-05-04:
Oh, actually I was wrong. It works with noetic, however I get the above error with kinetic or melodic docker files. Also, when I run all the launch files on our Linux machine with melodic ROS, I get the same error.

All the files including the docker compose files and the gh file is here: 
[all_files_plan_error.zip](https://github.com/compas-dev/compas_fab/files/11394905/all_files_plan_error.zip)

gonzalocasas, 2023-05-05:
Found and fixed on #383. A new release will come out soon including this fix.

begums, 2023-05-08:
Yes, it works now with older versions as well! Thanks a lot!

closed 2023-05-08 by begums