compas-dev/compas_fab #412: Add a doc page about user expectations

yck011522, 2024-04-12:
Level 0 (Complete Layman / No Code)
Visualization

Level 1
Define Robot Cell (Robot, Collision Environment, Tools, Workpieces)
Plan Motion
Plan Process

Level 2
Modify Robot Kinematics and Semantics (e.g. Planning Groups)
Custom Robot (e.g. multiple robots in cell)

Level 3 (Research level)
Extension with custom Planning Algorithms
Extension with custom Planning Backends

jf---, 2024-04-16:
+1

A nice way to approach this is perhaps through the use of the [`admonitions`](https://squidfunk.github.io/mkdocs-material/reference/admonitions/) module?

yck011522, 2024-04-16:
@jf--- 
This is new to me. Does this require extra work to setup the doc building process?

Maybe I will draft the content first.