compas-dev/compas_fab #426: Remove Legacy

yck011522, 2024-05-10:
I think it is time to remove the following 4 functions in favor of compas.Data. Especially for the compas_fab 1.0 release.

- `write_data_to_json`
- `read_data_from_json`
- `write_data_to_pickle`
- `read_data_from_pickle`

https://github.com/compas-dev/compas_fab/blob/main/src/compas_fab/utilities/file_io.py#L44-L111

Does anyone have reservations?

yck011522, 2024-05-29:
@gonzalocasas @chenkasirer @romanarust  ping.

Any thoughts about the removal of these?

gonzalocasas, 2024-05-31:
Agree! Probably `read_csv_to_dictionary` also needs to go, I don't think it's used currently anywhere in fab, and if it's not, it is not a good place for it as a generic util.