compas-dev/compas_fab #441: COMPAS FAB Rhino installation error

philianeles, 2024-09-12:
Rhino installation error:

(compas-dev) C:\Windows\System32>python -m compas_rhino.install                                                                          Traceback (most recent call last):                                                                                                         File "<frozen runpy>", line 198, in _run_module_as_main                                                                                  File "<frozen runpy>", line 88, in _run_code                                                                                             File "C:\Users\sercan\AppData\Local\anaconda3\envs\compas-dev\Lib\site-packages\compas_rhino\install.py", line 382, in <module>            install(version=args.version, packages=args.packages, clean=args.clean)                                                                File "C:\Users\sercan\AppData\Local\anaconda3\envs\compas-dev\Lib\site-packages\compas_rhino\install.py", line 52, in install              installation_path = compas_rhino._get_default_rhino_ironpython_sitepackages_path(version)                                                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^                                              File "C:\Users\sercan\AppData\Local\anaconda3\envs\compas-dev\Lib\site-packages\compas_rhino\__init__.py", line 512, in _get_default_rhino_ironpython_sitepackages_path                                                                                                             raise Exception("The default installation folder for rhino8 {} doesn't exist.".format(version))                                      Exception: The default installation folder for rhino8 8.0 doesn't exist.

philianeles, 2024-09-13:
@gonzalocasas would you have any leads on this that I could look into?

chenkasirer, 2024-09-13:
Hi @philianeles Rhino8 is currently the default installation location. This error seems to suggest the Rhino8 installation path doesn't exist.
Assuming you have Rhino7, could you try installing using:
```
python -m compas_rhino.install -v7.0
```

gonzalocasas, 2024-09-13:
If it's a fresh install of rhino 8, you have to open the legacy rhino script editor once and then run any python script at least once. That will create the folder.

Make sure you use the legacy component and not the new iron python 2 script component

philianeles, 2024-09-16:
@chenkasirer thank you! I switched to a completely new machine so just went ahead and installed Rhino 8, but @gonzalocasas thanks! yes, what you suggested seems to solve the problem, closing the issue.

closed 2024-09-16 by philianeles